mavlink/mavros #534: PX4.launch MAVROS crashes immediately

kchauvin, 2016-04-27:
Hello,

I am using a Raspberry Pi 3 running Raspbian Jessie and a Pixhawk.  Whether I install MAVROS from source or from binaries I always get this error, when running "roslaunch mavros px4.launch"

`[ INFO] [1461720576.061788025]: serial0: device: /dev/ttyAMA0 @ 57600 bps
pure virtual method called
terminate called without an active exception
================================================================================REQUIRED process [mavros-2] has died!
process has died [pid 28949, exit code -6, cmd /opt/ros/indigo/lib/mavros/mavros_node __name:=mavros __log:=/home/pi/.ros/log/7f5e37bc-0c17-11e6-99b0-b827eb7005ed/mavros-2.log].
log file: /home/pi/.ros/log/7f5e37bc-0c17-11e6-99b0-b827eb7005ed/mavros-2*.log
Initiating shutdown!
================================================================================
[mavros-2] killing on exit
`

Not sure if this is an issue with my build?  I was not able to build successfully when setting -DMAVLINK_DIALECT=common.  Not sure if this is a requirement with PX4.

Thank you

vooon, 2016-04-27:
Common dialect is base for all others, so ardupilotmega also support all messages from PX4.

What GCC version? What ROS distro used? For what platform code compiled: ARMv6, ARMv7 (armhf), ARMv8 (arm64/aarch)?

kchauvin, 2016-04-27:
GCC version is as follows:
gcc version 4.7.3 (Raspbian 4.7.3-11+rpi1) 

ROS version is ROS Indigo and armhf.

Thanks

vooon, 2016-04-27:
I can't find info what ABI used on Raspbian for Pi3. If they still use ARMv6 (Pi1 compatible) then you can't use ROS binaries (deb's).
ARM HF will not work properly with armel environment.

You should compile whole ROS from sources. Else you may try Ubuntu for Pi2 (not sure is v3 supported or not).

kchauvin, 2016-04-27:
@vooon Ok, I think it's also best to switch to Ubuntu at this point.  Unfortunately 14.04 LTS for Pi 2 does not run on Pi 3.  However Ubuntu Mate 15.10 does, and I am going to build ROS from sources as per this link:

http://userk.co.uk/ubuntu-15.10-install-ros-indigo-sources-chroot/

In terms of communication to PX4, does it matter if I am using Indigo or Jade?  I would like to stick with Indigo as I spoke to somebody who interfaced with the pixhawk before and said that it required Indigo at the time (last year).  Also would you recommend building ROS-desktop or just ROS-base?

vooon, 2016-04-27:
Indigo and Jade have same version of mavros. They mostly compatible.
Just year ago Indigo is a stable and Jade just released (not all packages officially available).
And in last year there was Hydro, where still slightly outdated mavros...

For onboard usage you usually do not need GUI packages, so robot base should be enought.
I do not see why to use chroot, but recommend to setup **distcc**, use USB-HDD for source files and use **catkin tool** instead of catkin_build_isolated.

kchauvin, 2016-04-28:
Firstly thanks for your help, I'm doing this for a school project and I'm new to ROS.  It's been a lot of effort trying to get this to build.  I decided instead to stick with Raspbian Jessie and Jade for now, I'm still running into a lot of bugs while building.  I followed the instructions and then I also installed missing dependencies as in the following issue.

https://github.com/mavlink/mavros/issues/508

I ended up needing about 29 other packages before rosdep would be ok and not have any OS conflicts.  Then when I built it all appeared ok, but when I ran rospack list, it did not show up and MAVROS appeared to not be installed.  

I started over again and rebuilt just MAVROS this time as the additional packages required by rosdep appeared to be installed.  It failed when trying to build MAVROS.  This was in the log file for MAVROS.  

Here is the output when I attempt the build.

http://pastebin.com/HvNKLcNg

What I think is the key is this error that causes it to stop the build:

'make[2]: *** No rule to make target '/usr/lib/liblog4cxx.so', needed by '/home/pi/catkin_ws/devel/.private/mavros/lib/libmavros.so'.  Stop.'

I've also attached the log file as well.  I get a lot of these warnings when building:

`CMake Warning at /home/pi/catkin_ws/src/mavros/mavros/CMakeLists.txt:92 (add_library):
  Cannot generate a safe runtime search path for target mavros_plugins
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libconsole_bridge.so] in /usr/lib/arm-linux-gnueabihf may be hidden by files in:
      /usr/local/lib

  Some of these libraries may not be found correctly.`

[build.make.000.log.zip](https://github.com/mavlink/mavros/files/239614/build.make.000.log.zip)

Thank you again.

vooon, 2016-04-28:
Do not forget to `catkin clean --all` after symlinking.

kchauvin, 2016-04-28:
Got it to work!  I followed these instructions and modified them slightly for Raspbian Jessie and Jade.  Seems to be working so far, just need to get it to talk with the Pixhawk now.

https://github.com/UAVenture/ros-setups/blob/master/intel-edison/install_ros_with_opencv.sh

closed 2016-04-28 by kchauvin