mavlink/mavros #538: Request failed. Check mavros log

Tondy, 2016-05-02:
After running: roslaunch mavros amp2.launch and successfully connected. 

I have tried arming the apm with : rosrun mavros mavsafety arm.

Result is: Request failed. Check mavros logs

Hope you can help.

vooon, 2016-05-02:
Please copy all text output (log) of console where you run roslaunch (a), then show diagnostics (b):
```
rostopic echo -n1 /diagnostics | tee /tmp/diag.yml
```

Also what wersion of ROS and Mavros?

Tondy, 2016-05-04:
Hello Voon

I am using  ros indigo. I will send text output when I run roslaunch. What will:
rostopic echo -n1 /diagnostics | tee /tmp/diag.yml Do ?

vooon, 2016-05-04:
`rostopic echo` prints all data from topic; [tee](http://man7.org/linux/man-pages/man1/tee.1.html).

closed 2016-06-23 by vooon

Tondy, 2016-07-16:
Hello

After running this command:
rostopic echo -n1 /diagnostics | tee /tmp/diag.yml

This message appears:

header: 
  seq: 2725
  stamp: 
    secs: 1468703221
    nsecs: 436976838
  frame_id: ''
status: 
  - 
    level: 1
    name: mavros: FCU connection
    message: not connected
    hardware_id: /dev/ttyUSB0:115200
    values: 
      - 
        key: Received packets:
        value: 0
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 0
      - 
        key: Tx sequence number:
        value: 138
      - 
        key: Rx total bytes:
        value: 0
      - 
        key: Tx total bytes:
        value: 1024105
      - 
        key: Rx speed:
        value: 0.000000
      - 
        key: Tx speed:
        value: 277.000000
  - 
    level: 2
    name: mavros: GPS
    message: No satellites
    hardware_id: /dev/ttyUSB0:115200
    values: 
      - 
        key: Satellites visible
        value: 0
      - 
        key: Fix type
        value: 0
      - 
        key: EPH (m)
        value: Unknown
      - 
        key: EPV (m)
        value: Unknown
  - 
    level: 2
    name: mavros: Heartbeat
    message: No events recorded.
    hardware_id: /dev/ttyUSB0:115200
    values: 
      - 
        key: Heartbeats since startup
        value: 0
      - 
        key: Frequency (Hz)
        value: 0.000000
      - 
        key: Vehicle type
        value: Generic
      - 
        key: Autopilot type
        value: Generic
      - 
        key: Mode
        value: ''
      - 
        key: System status
        value: Uninit
  - 
    level: 0
    name: mavros: System
    message: Normal
    hardware_id: /dev/ttyUSB0:115200
    values: 
      - 
        key: Sensor present
        value: 0x00000000
      - 
        key: Sensor enabled
        value: 0x00000000
      - 
        key: Sensor helth
        value: 0x00000000
      - 
        key: CPU Load (%)
        value: 0.0
      - 
        key: Drop rate (%)
        value: 0.0
      - 
        key: Errors comm
        value: 0
      - 
        key: Errors count #1
        value: 0
      - 
        key: Errors count #2
        value: 0
      - 
        key: Errors count #3
        value: 0
      - 
        key: Errors count #4
        value: 0
  - 
    level: 2
    name: mavros: Battery
    message: No data
    hardware_id: /dev/ttyUSB0:115200
    values: 
      - 
        key: Voltage
        value: -1.00
      - 
        key: Current
        value: 0.0
      - 
        key: Remaining
        value: 0.0
  - 
    level: 2
    name: mavros: Time Sync
    message: No events recorded.
    hardware_id: /dev/ttyUSB0:115200
    values: 
      - 
        key: Timesyncs since startup
        value: 0
      - 
        key: Frequency (Hz)
        value: 0.000000
      - 
        key: Last dt (ms)
        value: 0.000000
      - 
        key: Mean dt (ms)
        value: 0.000000
      - 
        key: Last system time (s)
        value: 0.000000000
      - 
        key: Time offset (s)
        value: 0.000000000

Tondy, 2016-07-16:
So

I run : roslaunch mavros px4.launch

then tried : rosrun mavros mavsafety arm

This become error : request failed. Check mavros logs.

I tried setting all pre arming check off.

vooon, 2016-07-17:
Did you see "level: 1 name: mavros: FCU connection message: **not connected**'?
What should ARM?

Tondy, 2016-07-17:
So the pixhawk board should ARM.

Tondy, 2016-07-17:
From Diagnostics command.

level: 1 name: mavros: FCU connection message: not connected hardware_id: /dev/ttyUSB0:115200 values:

vooon, 2016-07-17:
And how it may ARM if there is no connection?

Tondy, 2016-07-17:
So I have a usb serial connecting to the pixhawk telem2 pins with rx tx and ground.

Tondy, 2016-08-07:
Hello 

How can I Debug connection. Whats the correct way to guarantee a connection ? Or a link to a usb serial to pixhawk connection.  

Kind regards.

Tondy

vooon, 2016-08-07:
There are several tools for that. E.g. wireshark.

_// Are you on Mars? Such delays…_

Tondy, 2016-08-07:
Hello

Not yet on Mars. Only get to work on drone once in a while . I more of a coder. Will use wireshark
first debug that connection issue. Also if you have a helpful link please post.

Kind regards.

Tondy

vooon, 2016-08-07:
Try USB first. It should be `/dev/ttyACM%d` device. Check dmesg after plug. `minicom` may help to check tty communication.

APM as i know start mavlink by default, while PX4 give `nsh` shell. If it works then check params that set baud rate on TELEMn ports (check configuration manual, use MissionPlanner or QGC). Baud rate should match on both sides.

If params are ok: check your cable (connect Rx-Tx and use console), check that wire give correct cross: Rx1-Tx2, Tx1-Rx2, GND1-GND2.

Tondy, 2016-08-07:
Ok . Thank you very much.

Count-Darth, 2017-05-21:
hey everyone I am getting the same error, but my diagnostics are showing that everything else is fine can somebody help?

khancyr, 2017-05-22:
@Count-Darth Please open another issue with your information and log. If it failed to arm, it either you aren't correctly connected or you miss an arming check. Look at the message send back to see where the error is.