mavlink/mavros #540: How to test offboard control in Gazebo?

tom-xu-wen-bin, 2016-05-07:
http://dev.px4.io/ros-mavros-offboard.html 
I've followed this tutorial but I just can't figure out where I should put this cpp and compile. I want to test offboard control in Gazebo7. Any suggestion would be appreciated!

vooon, 2016-05-08:
http://wiki.ros.org/ROS/Tutorials

Starting from 3-rd item, for catkin.

tom-xu-wen-bin, 2016-05-08:
@vooon  Thank you very much, vooon. I've found that when I install mavros from source, I can create a catkin_ws folder and now I am figure out where I should put my cpp  in this folder and then catkin_make. Maybe I should review this tutorial about how to build a ros node. Another question, for mavros, is there a better tutorial than ReadMe? @vooon  Thank you very much, vooon. I've found that when I install mavros from source, I can create a catkin_ws folder and now I am figure out where I should put my cpp  in this folder and then catkin_make. Maybe I should review this tutorial about how to build a ros node. Another question, for mavros, is there a better tutorial than ReadMe?

tom-xu-wen-bin, 2016-05-08:
@vooon  Thank you very much, vooon. I've found that when I install mavros from source, I can create a catkin_ws folder and now I am figure out where I should put my cpp  in this folder and then catkin_make. Maybe I should review this tutorial about how to build a ros node. Another question, for mavros, is there a better tutorial than ReadMe?

Jaeyoung-Lim, 2016-06-09:
@WilliamUnderwood I guess you could check the mavros [ROS wiki](http://wiki.ros.org/mavros). Also there is a great comprehensive tutorial in http://dev.px4.io/simulation-ros-interface.html

closed 2016-06-23 by vooon