mavlink/mavros #542: distinguish arrived waypoint through mavros

CrazyPopLin, 2016-05-12:
Hi, I am just wondering if I get the informations that indicated the waypoints had been arrived from the rostopic "/mavros/mission/waypoints". 
For example: 4 waypoints in the list, I can find the first two mission items have been arrived. "is_current" is described what I am talking about?

vooon, 2016-05-12:
What do you mean on "arrived"? List is updated only after full transfer completion, except current flag (which anyway changes after recv).

`is_current` indicate that FCU execute this item.

CrazyPopLin, 2016-05-13:
I am using APM stack. 
The "arrived" means the waypoints have been went through or passed  by UAV.
*is_current* indicate the waypoint UAV flying to?
Thx.

vooon, 2016-05-13:
Yes, and at least APM do that (tested by me).

closed 2016-05-13 by vooon

CrazyPopLin, 2016-05-13:
Hi @vooon , I found **seq** is the one I need to figure out what is passed waypoint. So I change the **mavros_msgs/Waypoint** to add seq and also the **waypoint.cpp**.
but I got issue: 
>The dependencies of the message/service 'mavros_msgs/Waypoint' have changed. Please rerun cmake.  

How I build mavros after change it. Thx

vooon, 2016-05-13:
Why do you need it?! Siple array index is not enough?