mavlink/mavros #547: PX4 SITL command takeoff not taking off in a stable manner

ndepal, 2016-05-18:
I'm trying to control a drone (the iris model) in Gazebo through mavros. For this I have written a ROS node that sends commands/service requests on the `mavros/cmd/arming`, `mavros/set_mode`, `mavros/cmd/takeoff`, and `mavros/mission/...` to mavros.

I start the simulation with `make posix_sitl_default gazebo` and run mavros with `roslaunch mavros px4.lauch fcu_url:=udp://:14540@127.0.0.1:14557`, I can see that mavros can talk to PX4.

The problem that I have is that when I issue the command 

```c
ros::ServiceClient takeoff_client = ros_node_ptr_->serviceClient<mavros_msgs::CommandTOL>("mavros/cmd/takeoff");
        mavros_msgs::CommandTOL srv;
        srv.request.latitude = 0;
        srv.request.longitude = 0;
        srv.request.altitude = 10;
        srv.request.yaw = 0;
        srv.request.min_pitch = 0;

        if (!takeoff_client.call(srv)) {
            ROS_ERROR("Failed to takeoff.");
        } else {
            // still a failure
            if (!srv.response.success) {
                ROS_ERROR("Failed to takeoff (Error Code: %d)", srv.response.result);
            } else {
                ROS_INFO("Takeoff command sent");
            }
        }
```
The drone takes off and immediately pitches up, flying backwards at around 5 m/s. Sometimes it also scrapes the ground and flips.

If I just do `commander takeoff` in the PX4 shell, the drone takes off as expected.
I have the same problem when I try to take off with `rosrun mavros mavcmd takeoffcur 0 0 2`.

Is there something I am doing wrong?

Jaeyoung-Lim, 2016-06-08:
If you are running gazebo, I guess your initial local position (x,y,z)=(0,0,0) is not actually the (0,0,0) in the UTM coordinates( have you checked? )

Why not use the ~setpoint_position/local topic?

closed 2016-06-23 by vooon

LiYuanxing, 2017-09-19:
First, you have to go into the guided mode and take off.

TSC21, 2017-09-19:
> First, you have to go into the guided mode and take off.

@LiYuanxing wrong. First, PX4 does not have Guided mode. Second, does not have nothing to do with this issue.