mavlink/mavros #548: mavros_extras build error

liang-tang, 2016-05-20:
there is something i missed ? I pulled the leatest code and builded it, but something error with it.
Can anybody help me ?

build log:
Found '6' packages in 0.0 seconds. 
Starting ==> mavlink                                                                                                                  
Starting ==> mavros_msgs                                                                                                              
Finished <== mavros_msgs   [ 1.0 seconds ]                                                                                            
Finished <== mavlink       [ 25.7 seconds ]                                                                                           
Starting ==> libmavconn                                                                                                               
Finished <== libmavconn    [ 0.2 seconds ]                                                                                            
Starting ==> mavros                                                                                                                   
Finished <== mavros        [ 0.3 seconds ]                                                                                            
Starting ==> mavros_extras                                                                                                            
                                                                                                                                      
[mavros_extras] ==> '/home/liangtang/workspace/catkin_ws/build/mavros_extras/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' in '/home/liangtang/workspace/catkin_ws/build/mavros_extras'
[  9%] Built target servo_state_publisher
[ 18%] Built target copter_visualization
Linking CXX executable /home/liangtang/workspace/catkin_ws/devel/lib/mavros_extras/gcs_image_bridge
[ 90%] Built target mavros_extras
CMakeFiles/gcs_image_bridge.dir/src/gcs_image_bridge.cpp.o: In function `image_cb(boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&)':
gcs_image_bridge.cpp:(.text+0x106f): undefined reference to `cv::imencode(cv::String const&, cv::_InputArray const&, std::vector<unsigned char, std::allocator<unsigned char> >&, std::vector<int, std::allocator<int> > const&)'
CMakeFiles/gcs_image_bridge.dir/src/gcs_image_bridge.cpp.o: In function `cv::String::String(char const*)':
gcs_image_bridge.cpp:(.text._ZN2cv6StringC2EPKc[_ZN2cv6StringC5EPKc]+0x4f): undefined reference to `cv::String::allocate(unsigned long)'
CMakeFiles/gcs_image_bridge.dir/src/gcs_image_bridge.cpp.o: In function `cv::String::~String()':
gcs_image_bridge.cpp:(.text._ZN2cv6StringD2Ev[_ZN2cv6StringD5Ev]+0x14): undefined reference to `cv::String::deallocate()'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/liangtang/workspace/catkin_ws/devel/lib/mavros_extras/gcs_image_bridge] Error 1
make[1]: *** [CMakeFiles/gcs_image_bridge.dir/all] Error 2
make: *** [all] Error 2
[mavros_extras] <== '/home/liangtang/workspace/catkin_ws/build/mavros_extras/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' failed with return code '2'
                                                                                                                                       
Failed   <== mavros_extras [ 0.6 seconds ]                                                                                            
[build] There were '1' errors:                                                                                                        

Failed to build package 'mavros_extras' because the following command:

# Command to reproduce:
cd /home/liangtang/workspace/catkin_ws/build/mavros_extras && /home/liangtang/workspace/catkin_ws/build/mavros_extras/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j; cd -

# Path to log:
cat /home/liangtang/workspace/catkin_ws/build/build_logs/mavros_extras.log

Exited with return code: 2 
                                                                                                                                      
[build] Runtime: 26.9 seconds

vooon, 2016-05-20:
Check OpenCV installation. `sudo apt install libopencv-dev`

liang-tang, 2016-05-25:
OpenCV  installed, still compile error.

vooon, 2016-05-25:
Clean workspace and try again. Also check that cvimage bridge uses same opencv version.

And provide info on what OS used, version, ROS version.

vooon, 2016-05-25:
Check if that is useful: http://answers.ros.org/question/219640/reference-errors-after-opencv3-installation/

closed 2016-06-23 by vooon