mavlink/mavros #550: Setpoint Attitude Coordinate Frame Question

mtsakaguchi, 2016-05-26:
Hello,

I am using the mavros with the px4 SITL gazebo capability and have a question regarding the setpoint_attitude topic. When publishing a specified (roll,pitch,yaw) command to the setpoint_attitude topic, should I publish (roll,pitch,yaw) in relation to the ROS ENU frame and then mavros interprets these commands for the PX4 FCU in NED (similar to how setpoint_position commands are specified in ROS ENU and then mavros converts them) or do I publish (roll,pitch,yaw) commands directly in relation to the PX4 FCU in NED?

Any help is appreciated.

scott-eddy, 2016-06-02:
The euler angles sent over the ROS topic should describe the orientation of the base_link frame with respect to the ROS ENU frame.  [The appropriate frame conversions from base_link->aircraft and ENU->NED are done within MAVROS](https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/setpoint_attitude.cpp#L120-L122)

closed 2016-06-23 by vooon

chutsu, 2016-07-29:
So in short the setpoint attitude command should be in ENU frame?

scott-eddy, 2016-08-01:
Yes