mavlink/mavros #552: use usb port directly to pixhawk

YuYangguang, 2016-05-28:
I use a usb Android plug to connect to the pixhawk. But the issue was that it went wrong for it could not open the port. The reason was that I didnot use the serial port.

But I want to ask: Can I use this code to connect to the pixhawk using the usb-Android plug direcly? If so, how can I do this?  Can anybody solve this issue?

LorenzMeier, 2016-05-28:
I doubt you really ran mavros on Android. If you want to use Android you have to use the Java or C libraries for MAVLink.

closed 2016-05-28 by LorenzMeier

YuYangguang, 2016-05-28:
No, maybe I didn't explain it clearly right owing to the level of my English. I ran mavros on my notebook with Ubuntu14.04. I connect my computer and the pixhawk using an Android usb data line for I don't have the usb-to-serial data line right now. I want to send the control command from my  notebook. But it didn't work out as it reported that it could not open the port.
I'm not sure whether I have explained it clearly.  If you have got me, can you help me?

vooon, 2016-05-29:
I'm do not know what is "Android usb data line", some sort of USB cable (A-microB)?
If true, then you can use that cable to connect, check `/dev/ttyACM*` devices (and `dmesg` after cable plugging).

But that connection method not recommended to use in fly, rather then better to use USB-UART (FTDI) cable.