mavlink/mavros #555: waypoint receive issues

CrazyPopLin, 2016-06-08:
Hi, I found a issue when I tested a small project with mission planner and mavros together. I want the copter firstly fly with mission sent by mission planner, then sent new missions by mavros based on old missions if obstacle detected. 
I found mavros could receive missions only if mission planner sent missions before mavros started. And if mavros and mission planner already started, mavros couldn't receive new missions sent by mission planner. 
could you give me some suggestions. thx

vooon, 2016-06-23:
[pull_after_gcs][1] is set? Did you try `mavwp show -p`?

[1]: https://github.com/mavlink/mavros/blob/master/mavros/launch/px4_config.yaml#L111

CrazyPopLin, 2016-06-27:
yes, `pull_after_gcs` is set in **apm_config.yaml**. Also I tried `rosrun mavros mavwp show -p` and `rostopic echo /mavros/mission/waypoints`， no update after gsc write new missions in mission planner. I also tried AC3.3.1 and AC3.2.1, got same results. Any suggestions?

closed 2016-11-01 by vooon