mavlink/mavros #557: /mavros/setpoint_raw/local not work well with controlling velocity & position

rawk-v, 2016-06-09:
I am testing `/mavros/setpoint_raw/local` in gazebo.
I wan to control velocity when a target position set.

1. I have set all bit except `IGNORE_PX, IGNORE_PY, IGNORE_PZ, IGNORE_VX, IGNORE_VY, IGNORE_VZ` in message `PositionTarget` as explain in  [PositionTarget].(http://docs.ros.org/api/mavros_msgs/html/msg/PositionTarget.html)

2. When I set `IGNORE_PX, IGNORE_PY, IGNORE_PZ` too, velocity control works well, and when I set `IGNORE_VX, IGNORE_VY, IGNORE_VZ`, position control works well.

3. But when I want to control both velocity and position, **only** position control works, and the velocity is **always 3.0** no matter what I set to velocity.

This is really confuse me, because I think `/mavros/setpoint_raw/local` is use to control velocity and position simultaneously as #402 said.
Can any one help me? Thanks.

rawk-v, 2016-06-09:
@vooon 
PX4 v1.3.1 and mavros v0.17.3

rawk-v, 2016-06-09:
@vooon Thanks for replying, If this is a firmware side, can you tell me which version support this properties properly?

vooon, 2016-06-09:
Firmware receiver code: https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L693

And check setpoint loopback topic.

rawk-v, 2016-06-10:
Ok, I will check it, thanks @vooon .

TobiMiller, 2016-06-28:
same problem here. with some 2 month old px4 firmware/mavros and actual versions.

the question is. what should happen if i set position_z = 10 and velocity_z = 1 and the position is reached? is position placed over velocity in priority?

i think this is the problem:
https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L734

it ignores the whole group when one bit is set.

closed 2016-11-01 by vooon