mavlink/mavros #558: [ERROR] FCU: Initialising APM.

MohmadAyman, 2016-06-13:
hello guys,
i'm having a problem while connecting Apm2.6, ArduCopter 3.2.1 through mavros 0.17.3,
i get
`[ERROR] [1465812001.504243932]: FCU: ArduCopter V3.2.1 (36b405fb)
[ERROR] [1465812001.513742942]: FCU: Frame: QUAD`
the /mavros/state topic says its connected
`'connected: True
armed: False
guided: False
mode: STABILIZE'`
and when i try to arm it i get
```
[ERROR] [1465813087.476372944]: FCU: Calibrating barometer
[ERROR] [1465813087.485855288]: FCU: barometer calibration complete
[ERROR] [1465813092.523610963]: FCU: Initialising APM...
```

Is there a problem with the version or am i doing sth wrong,
does anyone have an idea ??

vooon, 2016-06-13:
It is not error! `'FCU: xyz'` messages come from autopilot, and older APM have incorrect severity levels.
So old mavros has quirk for that, but it removed since APM fixes severity levels.

MohmadAyman, 2016-06-13:
i see, thanks,
but the nth happens after the line 
`[ERROR] [1465813092.523610963]: FCU: Initialising APM...`
and the `  armed: False ` never changes, how can i fix that?

vooon, 2016-06-13:
Try to connect using APM Planner 2 (or Mission Planner) and check setup.
Anyway arming won't set until you command to arm.

MohmadAyman, 2016-06-13:
okay i'll check the setup again.

closed 2016-06-23 by vooon