mavlink/mavros #562: setpoint_velocity Trouble with linear z

Maladus, 2016-06-15:
I’m using the Arducopter 3.3 on PX4 Hardware. 
My Copter is the X8+ from 3DR.

When I send the velocity command over the rostopic setpoint_velocity/cmd_vel to the Copter
X- and Y-Axis work well, but the Z-Axis and Z-Orientation doesn‘t work.

While commands on y- and y- axis result in a smooth, correct motion, commands on the z-axis lead to a sudden jump in z-velocity and therefore height. The copter does not climb or fall smoothly. However, the overall z-command seems to be O.K., as the robot moves down or up when sending the according command.

Why it doesn‘t it work, or what is my mistake?

Jaeyoung-Lim, 2016-07-19:
My opinion is that it comes from the inaccuracy of the altitude information.

If you are using only the barometer for altitude estimation, the error of the altitude will be somewhat course. Try adding a lidar sensor for altitude measurements.

Maladus, 2016-07-19:
Yes, I have only the barometer for altitude estimation used.
Unfortunately I can't tested, because I had to return the hardware.
I had it only for a short time.

closed 2016-07-19 by Maladus