mavlink/mavros #563: Tegra TK1 multiple UART problem

scott-eddy, 2016-06-21:
*Disclaimer this problem is* **extremely** *configuration dependent, I don't expect an answer from the community but I would appreciate any guidance people could give.*

I am using the [Tegra TK1](http://elinux.org/Jetson_TK1) with the [grinch 21.3.4 kernel](https://devtalk.nvidia.com/default/topic/906018/jetson-tk1/-customkernel-the-grinch-21-3-4-for-jetson-tk1-developed/) to communicate to a pixhawk running the PX4 stack.  Mavros's standard use case of communicating between the boards works perfect over the USB (/dev/ttyACM0) and the serial ports (/dev/ttyTHS*) of the Tegra.

I've attached a serial range sensor to the Tegra, the low-level I/O is handled by the [serial ros package developed by wjwwood.](https://github.com/wjwwood/serial)  Again, the standard use case of reading in my sensor's range data works wonderfully.  

However, here's the interesting part.  When running mavros and my serial_sensor node on serial ports /dev/ttyTHS0 and /dev/ttyTHS1 (doesn't matter the order), the range data reported by serial_sensor is lagged in time [will post images to detail this in the morning].  This problem occurs even if I flush the UART's internal buffers via [serial.flush()](http://wjwwood.io/serial/doc/1.1.0/classserial_1_1_serial.html#a63b7abf172cad25bfc998b3b1f98310f)  However, if I run mavros on the USB port (/dev/ttyACM0) and serial_sensor on the UART port (/dev/ttyTHS0) the serial_sensor data tracks range in real time.

Since both packages can run on the CPU in real time without conflict this appears to me that their threads are fighting over a UART resource.  **Note: I do not understand why this would be as the serial ports show up in the device tree under different names (e.g. ttyTHS0 and ttyTHS1)** 

Not being one to shy away from details I dove a little into how the communications are handled in both packages.  Mavros, via libmavconn, uses [boost::asio and std::lock_guard](https://github.com/mavlink/mavros/blob/master/libmavconn/src/serial.cpp#L25-L39) in order to address the issue of low-level I/O and thread safety.  Serial on the other hand uses [pthreads for thread safety](https://github.com/wjwwood/serial/blob/master/include/serial/impl/unix.h#L211-L214) [and std::unistd](https://github.com/wjwwood/serial/blob/master/src/impl/unix.cc#L545) for low-level I/O.

I'm looking for guidance on the following:

1.  Does anyone know if libmavconn's (lock_guard) (https://github.com/mavlink/mavros/blob/master/libmavconn/src/serial.cpp#L29) should ever unlock?  During writes it calles lock but it never unlocks.

2. Does anyone know if boot::asio and pthreads can play nice together?

3. Does anyone know if two UARTs in the device tree could be the same mutex at the C++ level?

4. Anyone have any advice for what to explore?

Any advice greatly appreciated.  

Thanks, 
-Eddy

vooon, 2016-06-21:
1) [std::lock_guard][lg] - automatically handle lock and unlock of passed [std::mutex][mu] (or `std::recursive_mutex`). And on Linux `libstdc++` uses pthread for threading.

2) `boost:asio` also should use `::read()` and `::write()`, but different poll: `pselect()` on serial, and various on asio (e.g. `epoll()`).

3) Check addresses in DT: should be different for UART IP-cores, but they should be ok - if i remenber correctly linux forbid several remap to same of_resource.

If you could try to connect two FCU's (or same FCU by two telemetry channels) to `ttyTHS0`, `ttyTHS1` and start two instances of mavros:

```
rosrun mavros __ns:=ths0 _fcu_url:=/dev/ttyTHS0 _gcs_url:=""
rosrun mavros __ns:=ths1 _fcu_url:=/dev/ttyTHS1 _gcs_url:=""
```

Also use `strace` to check polling.

[lg]: http://en.cppreference.com/w/cpp/thread/lock_guard
[mu]: http://en.cppreference.com/w/cpp/thread/mutex

scott-eddy, 2016-06-21:
Thanks for the reply voon.  I'm going to be using this issue to try and trace down the problem so that way it might serve to help others in the future, I will close it either when a solution is found or when I abandon the effort.

3`)  in /proc/device-tree of the TK1 there are the following serial ports:
- `serial@70006000` == uarta: reg <0x0 0x70006000 0x0 0x40> [on tegra124-soc.dtsi]
- `serial@70006040` == uartb: reg <0x0 0x70006040 0x0 0x40>
- `serial@70006200` == uartc: reg <0x0 0x70006200 0x0 0x40>
- `serial@70006300` == uartd: reg <0x0 0x70006300 0x0 0x40>

So each serial line should indeed be a different resource

scott-eddy, 2016-06-21:
Appending my launch file with an strace for the serial_sensor node and testing with MAVROS on the USB and UART yields the following:

** MAVROS on USB Port **
% time     seconds  usecs/call     calls    errors syscall
------ ----------- ----------- --------- --------- ----------------
 46.28    0.070000          68      1024           pselect6
 24.07    0.036402          56       652           futex
 13.50    0.020422          34       598           nanosleep
  3.61    0.005453           1      4284           clock_gettime
  3.42    0.005175           2      3010           ioctl
  2.94    0.004448           4      1165           write
  1.77    0.002676           1      2583       140 read
  1.35    0.002037          10       203           select
  1.10    0.001664           9       181           close
  1.05    0.001590          24        66        66 connect
  0.53    0.000808           3       292       188 open
  0.07    0.000111           1       146        82 stat64
  0.06    0.000087           1        66           mprotect
  0.06    0.000086           1       102           fstat64
  0.05    0.000082           1        73           socket
  0.05    0.000079           0       165           mmap2
  0.04    0.000066           1        70           munmap
  0.03    0.000043           2        26        24 access
  0.02    0.000023           0        72           fcntl64
  0.00    0.000000           0         1           execve
  0.00    0.000000           0        62           lseek
  0.00    0.000000           0         1           pipe
  0.00    0.000000           0         3           brk
  0.00    0.000000           0        25           gettimeofday
  0.00    0.000000           0         4           clone
  0.00    0.000000           0         5           uname
  0.00    0.000000           0         4           rt_sigaction
  0.00    0.000000           0         1           rt_sigprocmask
  0.00    0.000000           0         1           getrlimit
  0.00    0.000000           0         1           set_tid_address
  0.00    0.000000           0         5           bind
  0.00    0.000000           0         2           listen
  0.00    0.000000           0         7           getsockname
  0.00    0.000000           0         4           sendto
  0.00    0.000000           0         3           shutdown
  0.00    0.000000           0         5           setsockopt
  0.00    0.000000           0        12           recvmsg
  0.00    0.000000           0         1           set_robust_list
  0.00    0.000000           0         1           set_tls
------ ----------- ----------- --------- --------- ----------------
100.00    0.151252                 14926       500 total

** MAVROS on UART Port **
% time     seconds  usecs/call     calls    errors syscall
------ ----------- ----------- --------- --------- ----------------
 42.28    0.068822          25      2778         1 futex
 30.83    0.050185          61       818           nanosleep
 12.90    0.021000          85       246           pselect6
  4.24    0.006899           4      1872           write
  2.36    0.003837           2      2230           ioctl
  2.33    0.003794           1      4932           clock_gettime
  1.85    0.003017           2      1802       137 read
  1.31    0.002130          12       181           close
  0.86    0.001400          21        66        66 connect
  0.61    0.001000           5       200           select
  0.16    0.000258           0       736           gettimeofday
  0.12    0.000202           1       146        82 stat64
  0.05    0.000080           1        66           mprotect
  0.05    0.000076           0       292       188 open
  0.03    0.000045           0       165           mmap2
  0.02    0.000028           0       102           fstat64
  0.01    0.000016           0        72           fcntl64
  0.00    0.000000           0         1           execve
  0.00    0.000000           0        62           lseek
  0.00    0.000000           0        26        24 access
  0.00    0.000000           0         1           pipe
  0.00    0.000000           0         3           brk
  0.00    0.000000           0        70           munmap
  0.00    0.000000           0         4           clone
  0.00    0.000000           0         5           uname
  0.00    0.000000           0         4           rt_sigaction
  0.00    0.000000           0         1           rt_sigprocmask
  0.00    0.000000           0         1           getrlimit
  0.00    0.000000           0         1           set_tid_address
  0.00    0.000000           0        73           socket
  0.00    0.000000           0         5           bind
  0.00    0.000000           0         2           listen
  0.00    0.000000           0         7           getsockname
  0.00    0.000000           0         4           sendto
  0.00    0.000000           0         3           shutdown
  0.00    0.000000           0         5           setsockopt
  0.00    0.000000           0        12           recvmsg
  0.00    0.000000           0         1           set_robust_list
  0.00    0.000000           0         1           set_tls
------ ----------- ----------- --------- --------- ----------------
100.00    0.162789                 16996       498 total

Of note is pselect6 goes from 1000 calls to 200, and futex goes from 652 calls to 2778

vooon, 2016-06-21:
Time under strace may be different or USB settings use higher message rates.

And be sure to use `-f` flag, else IO-thread will be missed.

I don't see anything suspicious in this summaries. Could you try use two asio-based serial test programs?
Make cross-connection of UARTs then do ping. Since both programs run on same computer you may use `gettimeofday()` for monotonic timestamp.

closed 2016-06-22 by scott-eddy

vooon, 2016-06-22:
@scott-eddy also you may test [ttyping.cc][1].

```
./ttyping /dev/ttyTHS0 --pong
./ttyping /dev/ttyTHS1
```

[1]: https://gist.github.com/vooon/e50fd7d38c137aa932b8caf18326d12e