mavlink/mavros #566: Serial receive: End of file

khancyr, 2016-06-28:
Hi,

Sometimes I got : 

````
[ INFO] [1467126122.716706744]: serial0: device: /dev/ttyACM0 @ 115200 bps
[ INFO] [1467126122.717268186]: GCS URL: udp://:6001@localhost:14555
[ERROR] [1467126122.717306127]: serial10947712: receive: End of file
[ INFO] [1467126122.717668851]: udp1: Bind address: 0.0.0.0:6001
[ INFO] [1467126122.717781511]: udp1: Remote address: 127.0.0.1:14555
terminate called after throwing an instance of 'std::system_error'
  what():  Resource deadlock avoided
================================================================================REQUIRED process [mavros-2] has died!
process has died [pid 17733, exit code -6, cmd /home/khancyr/sncf_ws/devel/lib/mavros/mavros_node __name:=mavros __log:=/home/khancyr/.ros/log/4508b29e-3d41-11e6-9eaa-40e230b7507d/mavros-2.log].
log file: /home/khancyr/.ros/log/4508b29e-3d41-11e6-9eaa-40e230b7507d/mavros-2*.log
Initiating shutdown!
````

Only a reboot solve the issue....
The issue was present on previous release.
But something change since you update libmavconn (congrate by the way ! seems working but with lot of new warning)
Previously I only got 
serial0: receive: End of file
Now I got serial10947712: receive: End of file and the number is changing every try : serial11056608: , serial19591504: etc

I try to see what is wrong in libmavconn but find nothing wrong... 
Any clues ?

vooon, 2016-06-28:
Number "bug" fix - 8a1de7ce287af951363ea24a6a70aa74cae23a4b

But that couldn't solve EOF. I'm seen this too and only reboot helps. No clue why.

vooon, 2016-06-28:
Oh, i often seen EOF after `mavproxy` usage, HW-reconnect won't help.

mechy45, 2016-10-04:
Check your input voltage of pixhawk power. 4.99 volt  makes "serial0: receive : End of file" error in my case. 5.01 volt doesn't matter.

closed 2016-10-04 by vooon

mechy45, 2016-10-07:
This issue has been proved to be a looseness of the usb connector in my case.
I connected using a usb cable between Intel NUC and usb to serial converter which connected with Pixhawk TELEM2. 
When I shake a usb cable which connected with NUC slightly, serial0: error occurs every time.

xuyuan, 2016-10-11:
Hi, I got the same problem when pixhawk is powered from USB, however QGroundControl can connect to pixhawk without any problem, so I think this is bug

pavloblindnology, 2017-11-17:
Found solution without restarting a companion computer [#865 (comment)](https://github.com/mavlink/mavros/issues/856#issuecomment-345217529).

RoboticsPolyu, 2023-12-01:
I also encountered this issues. It looks like a bug. Because after the app exited,  the app can be started again.

khancyr, 2023-12-01:
this issue was solve long time ago. if there is a new, you should do a full report of your system and how the issue trigger