mavlink/mavros #569: python: Update mavros.mavlink module for 2.0 support.

vooon, 2016-06-28:


vooon, 2016-07-03:
Module updated for 2.0, but without signing in `convert_to_rosmsg()` since i do not found how extract signature block.

closed 2016-07-03 by vooon

danielkalmanson, 2024-04-16:
Hi,
The issue with the signature extraction on mavlink2.0 to ros mavlink makes my code crash every time with the following error message:
> File "/opt/ros/humble/local/lib/python3.10/dist-packages/mavros/mavlink.py", line 117, in convert_to_rosmsg return Mavlink(
> File "/opt/ros/humble/local/lib/python3.10/dist-packages/mavros_msgs/msg/_mavlink.py", line 174, in init self.signature = array.array('B', kwargs.get('signature', []))
> TypeError: 'NoneType' object is not iterable
> The following exception was never retrieved: 'NoneType' object is not iterable

I believe the problem is that the signature argument given to the message is always None, while the message builder in _mavlink.py expects an array.
Even without the full fix of extracting Mavlink signatures properly, I think a simple [] instead of None given to the message, could be a quick patch to make the code run smoothly.

Would very much appreciate this fix!

@vooon

danielkalmanson, 2024-04-17:
Yes, that's what I did in my code, but I think it would be a great idea to push it to the master branch, so people would have this fix built-in when downloading or updating the package.
I wouldn't want to fix this issue every time I update the package from the source.

vooon, 2024-04-17:
You're welcome to PR.