mavlink/mavros #570: Creating OSD node

lefos99, 2016-07-01:
Hello,
I am trying to create a node, which will communicate with the osd by serial (serial:///dev/ttyUSB1)...The communication seems ok, I have created _mavconn::MAVConnInterface::Ptr_ object and I am able to send the data to the OSD...
My issue is that I cannot kill the node and it seems that is because this pointer! By the way I do not know exactly how to close properly the link I create...(I have used the close() function but it doesn't seem to work...). 

```
int main(int argc, char **argv)
{
    ros::init(argc, argv, "mavros_osd");
    ros::NodeHandle osd_node;
    diagnostic_updater::Updater updater;
    std::string osd_url = "serial:///dev/ttyUSB1"; //TODO check if the path is correct
    ROS_INFO_STREAM("OSD URL: " << osd_url);
    updater.setHardwareID(osd_url);
    try
    {
        osd_link = mavconn::MAVConnInterface::open_url(osd_url,1, MAV_COMP_ID_OSD); //TODO change it!
        osd_link_diag.set_mavconn(osd_link);
        if (osd_link)
        {
            osd_link_diag.set_connection_status(true);
        }
        updater.add(osd_link_diag);
    }
    catch (mavconn::DeviceError &ex)
    {
        ROS_FATAL("OSD: %s", ex.what());
        ros::shutdown();
        return -1;
    }

    // updater spinner
    auto diag_timer = osd_node.createTimer(ros::Duration(0.5),
            [&](const ros::TimerEvent &evt) {
                updater.update();
            });
    diag_timer.start();

    MavrosOSD mavrosOSD(osd_node);
    diag_timer.start();
    while(ros::ok())
    {
    // doing stuff
    ros::spinOnce();
    }
    return 0;
}
```

Any clues ?

vooon, 2016-07-01:
What version of libmavconn? Old one have bug #130, fixed in 0.18 by #527.

lefos99, 2016-07-01:
`dpkg -l | grep libmavconn` gave me 

> ros-jade-libmavconn: 0.16.6-0trusty

So should I try to update from source file...? That means that the repository is outdated?

vooon, 2016-07-01:
Jade and Indigo support is separated out, but master (kinetic) still builds (travis ci).

#527 is in 0.17.3+, so you can use older release. Or use master, for that be sure to set `--rosdistro kinetic` at rosinstall_generator.

lefos99, 2016-07-01:
Right now I am using ros-jade in my project and according to [wiki of mavros](wiki.ros.org/mavros), jade is using the branch: indigo-devel, which is in 0.17.4...So can I just install the lib by using `--rosdistro jade` at rosinstall_generator?

vooon, 2016-07-01:
Yes, you can. But 0.18 has different API because MAVLink 2.0 support.

lefos99, 2016-07-04:
Yes ,that seems to solve the problem! Thanks! :+1:

closed 2016-07-04 by lefos99