mavlink/mavros #575: Help with Rover - GUIDED mode not responding to setpoint_position with mavros

botjunkie, 2016-07-16:
Hi guys,

I need to do use the GUIDED mode on APMRover2 using mavros. Basically, I want to give GPS co-ordinates of a location to the rover (through a script, not AUTO mode) and make the rover and go and stay there for a while, and then make it go to another location. I have done a similar script for COPTER and it works just fine. I have tested that out in SITL and on the real platform. However, when I set > setpoint_position (SET_POSITION_TARGET_LOCAL_NED) from the script (via mavros) the ROVER does not respond.

I am able to put the ROVER in guided mode and arming it before doing anything else. But setpoint_position just doesnt work.

I have also tried using mavproxy and it works just fine. I can change the mode to GUIDED, arm the ROVER and then click on a location on the map and it goes there. I can also override RC and have direct PWM control of throttle and steering. Having said that, I dont want to control PWM from my script. Instead I want to give the autopilot a GPS location and make it figure out the PWM commands needed to get there.

I have checked some of the related topics and tried changing parameters.
ARMING_CHECK 0
ARMING_REQUIRE 0
SYSID_MYGCS 1
But the rover still wont respond to the script.

Can you please help?

I hope I made it clear enough.

vooon, 2016-07-17:
https://github.com/ArduPilot/ardupilot/issues/4488 identical issue.

I do not see any `SET_.*_TARGET` in `APMrover2/GCS_Mavlink.cpp` (and see in plane and copter).
So APM:Rover simply do not support that messages. Only one possible way is non standard usage of `MISSION_ITEM` in GOTO mode.

But i dropped goto support in last year. Write special plugin or switch to dronekit.

khancyr, 2016-07-17:
Only setpoint with GPS coordinate work on Rover.
To use MAVROS you only need SYSID_MYGCS 1. Don't set the other at 0

botjunkie, 2016-07-17:
Thanks for your comments guys. 

@vooon You are right, APMRover does not support SET_POSITION_TARGET_LOCAL_NED and other setpoint messages. I dont want to switch to dronekit. I want to continue with mavros. What exactly should I write the plugin for? mavros already has a plugin for _setpoint_position_, do you mean that I should add/modify on the APMRover side? If I were to write a plugin, can you provide me a link where I can begin from? I know the mavros plugins, but I would like to simplify the problem and understand it better. Thanks for your help!

@khancyr 

> Only setpoint with GPS coordinate work on Rover.

How? Do you have an example script? Do you mean it works in GUIDED mode? It seems like none of the _setpoint_ messages are supported in ROVER as @vooon pointed out. How do you convey the GPS coordinates? Do you load the waypoints and switch to AUTO or do you use _setpoint_raw_(mavros)? Does _setpoint_raw_ work? I want to be able to do this from a script. 
Please let me know. 

Thanks!

vooon, 2016-07-17:
@botjunkie `setpoint_raw` uses same messages as other `setpoint_xxx` plugins. It can't help if you do not has handler in firmware.

So i see two possible solutions:
a) Add support to firmware side. See how copter/plane handle it and do similar. At least `SET_POSITION_TARGET_GLOBAL_INT` should be easy to add because it very close to GOTO.

b) Write plugin to send GOTO commands. APM uses `MISSION_ITEM` with magic value `2` for that.
You may find it in dronekit: [dronekit.Vehicle.simple_goto()](https://github.com/dronekit/dronekit-python/blob/master/dronekit/__init__.py#L1965-L2015)

welting82, 2016-07-19:
I have same problem as you few month ago. And,I find out those be deleted code on the history of waypoit.cpp and mavwp.py(GOTO service).it's working after I modify those code.
So,you can try it.

vooon, 2016-07-19:
@welting82 goto removed because several reasons: it is non standard, not related to mission protocol, there standardized SET_POSITION_TARGET_GLOBAL_INT, ROS Service is bad idea for setpoint.

closed 2016-11-01 by vooon

pzxz, 2021-09-12:
@botjunkie I have the same problem as you, have you solved the problem?