mavlink/mavros #576: MAVROS raspberry pi installation error

Jaeyoung-Lim, 2016-07-19:
I am trying to install mavros on Raspberry Pi 2

Following the [instructions](https://github.com/mavlink/mavros/tree/master/mavros) of installing mavros from source, I get an error from
```
rosdep install --from-paths src --ignore-src --rosdistro indigo --os=debian:jessie -y
```

[image]

How can I solve this problem?

vooon, 2016-07-19:
Did you install ROS? I mean base system.

Check that guide: http://dev.px4.io/ros-raspberrypi-installation.html

vooon, 2016-07-19:
Also note that debian officially supported starting from Kinetic.

Jaeyoung-Lim, 2016-07-21:
@vooon I realized that I didn't install the released packages at 4.2
However, I am not being able to install it properly.

Are you suggesting that it is better to install kinetic than indigo? I supposed it was better to follow the tutorial as indigo

Jaeyoung-Lim, 2016-07-21:
I have reinstalled raspbian(Jessie) and followed the whole process of the guide: http://dev.px4.io/ros-raspberrypi-installation.html

Still have the same error message of not being able to find package for OS [debian]

vooon, 2016-07-21:
Switch to kinetic may help, but may don't. Now i don't have access to my RPi2, but i used Ubuntu image.

Jaeyoung-Lim, 2016-07-21:
So I had to use rosinstall_generator to get a bunch of packages which rosdep could not find. These are the packages I had to add to the mavros.install file.
```
rosinstall_generator geometry_msgs | tee -a /tmp/mavros.rosinstall
rosinstall_generator sensor_msgs | tee -a /tmp/mavros.rosinstall
rosinstall_generator tf | tee -a /tmp/mavros.rosinstall
rosinstall_generator control_toolbox | tee -a /tmp/mavros.rosinstall

rosinstall_generator realtime_tools | tee -a /tmp/mavros.rosinstall
rosinstall_generator nav_msgs | tee -a /tmp/mavros.rosinstall
rosinstall_generator urdf | tee -a /tmp/mavros.rosinstall
rosinstall_generator eigen_conversions | tee -a /tmp/mavros.rosinstall
rosinstall_generator  | tee -a /tmp/mavros.rosinstall
rosinstall_generator urdf | tee -a /tmp/mavros.rosinstall

rosinstall_generator actionlib_msgs | tee -a /tmp/mavros.rosinstall
rosinstall_generator dynamic_reconfigure | tee -a /tmp/mavros.rosinstall
rosinstall_generator diagnostic_msgs | tee -a /tmp/mavros.rosinstall
rosinstall_generator pluginlib | tee -a /tmp/mavros.rosinstall
rosinstall_generator orocos_kdl | tee -a /tmp/mavros.rosinstall
rosinstall_generator visualization_msgs | tee -a /tmp/mavros.rosinstall
rosinstall_generator angles | tee -a /tmp/mavros.rosinstall
rosinstall_generator tf2_py | tee -a /tmp/mavros.rosinstall
rosinstall_generator tf2 | tee -a /tmp/mavros.rosinstall
rosinstall_generator urdf | tee -a /tmp/mavros.rosinstall

rosinstall_generator diagnostic_updater | tee -a /tmp/mavros.rosinstall
rosinstall_generator image_transport | tee -a /tmp/mavros.rosinstall
rosinstall_generator actionlib | tee -a /tmp/mavros.rosinstall
```

However, rosdep was showing unresolved dependencies
```
executing command [sudo -H apt-get install -y liburdfdom-dev]
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 liburdfdom-dev : Depends: libconsole-bridge-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y liburdfdom-dev] failed
```

Anyway, catkin build is not being able to build with mavros a weird messsage in roscpp
[image]

annesteenbeek, 2016-07-28:
"fixed" this by adding --os=ubuntu:trusty

`rosdep install --from-paths src --ignore-src --rosdistro indigo --os=ubuntu:trusty -y`

was allso missing liburdf packages for mavros-extras, got these from the ubuntu trusty armhf ppa. Installing mavros using binaries still causes problems for me on the RPI3 which kinda sucks since building it from source takes hours.

Jaeyoung-Lim, 2016-07-29:
@annesteenbeek   have you done the rosdep on raspbian Jessie? It's good if it works!

I have fixed the problem by changing running [ubuntu 14.04](https://wiki.ubuntu.com/ARM/RaspberryPi) on Raspberry Pi

annesteenbeek, 2016-07-29:
Yes, raspbian jessie lite on RPI3, it did give some errors when starting up,  (don't have access to rpi atm else I would post them)  but it ran fine. Would like to have installed ubuntu but couldn't find any 14.04 images for RPI3. Do think the wiki tutorial and PX4 tutorial should be updated since both don't work as step by step guides anymore.

ChrisYin, 2017-06-06:
@annesteenbeek hello! I had the same problem in RPI3 when using rosdep. Have you solved this problem? or should I change system into ubuntu 14.04 in RPI3?

annesteenbeek, 2017-06-06:
If I remember correctly, I solved this by adding some u until repositorys. However. I do think the easiest method to solve this is to run Ubuntu on the pi. Be it 14.04 or 16.04. This also helps with overall Ros comparability.

Jaeyoung-Lim, 2017-06-07:
@ChrisYin In my opinion, changing to ubuntu 14.04 will save you a lot of time. Raspian will keep having compatibility issues using different ROS packages

chngdickson, 2018-02-09:
@Jaeyoung-Lim Hello kind sir, have u actually got it up and running? If so, can u please share me ur solution for this matter?

Jaeyoung-Lim, 2018-02-09:
@chngdickson I ended up giving up on Raspbian and installed Ubuntu 14.04 on RPi2
I think this is a better solution and saves a lot of trouble using other ROS packages too