mavlink/mavros #577: Can't read mavros/rcin

zer1988, 2016-07-20:
Hi everbody,
I am going to receive the roll pitch yaw and throttle from my radio transmitter using RCin topic of Mavros. My problem is that I do not know how I can read each channels individually. Everytime I program like msg.channels[0] for example. My FCU reclaims that process has died.
Thanks.

vooon, 2016-07-20:
Did you check `channels.size()` before accessing?

zer1988, 2016-07-20:
Oh I did not do that. I used command "rostopic echo /mavros/rc/in" and i saw like this 
header: 
  seq: 460
  stamp: 
    secs: 1456951630
    nsecs: 476419853
  frame_id: ''
rssi: 0
channels: [982, 982, 900, 900, 900, 900, 900, 900]

But when I used this function to get the value from radio transmitter
void rcin_call(const sensor_msgs::Imu::ConstPtr& msg){
    rc_msg = *msg;
}
and check the value of throttle that is channel 3
ROS_INFO( " Value throttle: %d", rc.channels[3]). 

When I run my launch file. It shows me one error that the process has died and rqt_graph is only shows mavros topic and tf topic.
Thanks.

vooon, 2016-07-20:
Guess what happen here:
```cpp
#include <vector>

int main()
{
	std::vector<int> v;

	v[0] = 1;

	return 0;
}
```

zer1988, 2016-07-20:
Hi vooon,
In this morning, I tested with this command
ROS_INFO( " Value throttle: %.8f", (double) msg->channels[0])
and it works!.
But i would like to know why it can run.with this one.
Thanks.;

closed 2016-07-29 by zer1988

Guiboy, 2017-03-04:
Hi 
I am very new to ROS and would like to read my APM RC values from a python script using MAVROS node.
I have a running Mavros node and can display the RC channel values using the "rostopic echo /mavros/rc/in" command.

Could you please share the complete code where you integrate "ROS_INFO( " Value throttle: %.8f", (double) msg->channels[0])"?
I know nothing about ROS programming and would like to reuse the exact same function.

Thanks in advance!
Guillaume

vooon, 2017-03-06:
Something like that:

```python
import rospy
import mavros_msgs

# see ROS tutorials! init required
rospy.init_node("rcin_sub")

sub = rospy.Subscriber("mavros/rc/in", mavros_msgs.RCIn, lambda msg: print(msg))

rospy.spin()
```