mavlink/mavros #579: Problem installing MAVROS on ODROID C2 running Ubuntu 16.04 Xenial

siddhu45, 2016-07-26:
Hello Community,
I am trying to source install MAVROS on ODROID C2 running Ubuntu 16.04 Xenial but am having errors while building the workspace. I have installed ROS Indigo and it works just fine. When I try to install dependencies using the rosdep command I get the following error
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
mavros_extras: No definition of [cv_bridge] for OS version [xenial]
test_mavros: No definition of [control_toolbox] for OS version [xenial]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```
The error while running catkin_make is:
```
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "image_transport"
  with any of the following names:

    image_transportConfig.cmake
    image_transport-config.cmake

  Add the installation prefix of "image_transport" to CMAKE_PREFIX_PATH or
  set "image_transport_DIR" to a directory containing one of the above files.
  If "image_transport" provides a separate development package or SDK, be
  sure it has been installed.
Call Stack (most recent call first):
  mavros/mavros_extras/CMakeLists.txt:7 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/odroid/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/odroid/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:290: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
```
If anyone can help me would be a great help! 
Thanks

vooon, 2016-07-26:
According to [REP 3][1] you should use Kinetic for Ubuntu 16.04. Next whay you not follow instructions: [README.md][2]?

[1]: http://www.ros.org/reps/rep-0003.html#kinetic-kame-may-2016-may-2021
[2]: https://github.com/mavlink/mavros/blob/master/mavros/README.md#installation

siddhu45, 2016-07-27:
Thank you so much for the reply! I know that I should be able to install ROS Kinetic but due to some errors I'm not able to do that! I think the error is related to QT5 somehow.
And for the mavors, yes, I followed the very same instructions to install mavros from source but faced errors as mentioned above. Could not determine the OS to get required dependencies I guess.
Any further comments on that, please?

Thanks

jgoppert, 2016-07-27:
I would try to fix the qt5 issue and use kinetic. The only other way is to build all of the ros packages from source, which is doable, but not fun

vooon, 2016-07-27:
You can build that packages from source, but that not best way. Better to find out how to fix kinetic.

That is example, but i think that it will require additional packages.
```
rosinstall_generator --rosdistro indigo cv_bridge image_transport | tee -a /tmp/dep.ri
wstool merge -t src /tmp/dep.ri
```

siddhu45, 2016-07-28:
@jgoppert I will try to build ros packages individually from source for kinetic. -Thanks
@vooon I tried building mavros from source but couldn't run `catkin_make`. Also what do you mean by fixing kinetic? How does it work. Can you explain in more detail and steps on how to solve that issue!

And has anyone installed mavros on ubuntu 16.04 Xenial as its status still shows "build failing" .. they are trying to use docker to install mavros on ubuntu 16.04 Xenial. Any idea how to use that and install -Thanks

vooon, 2016-07-28:
That build failed because it uses one step older mavlink (2016.6.24 instead of 2016.7.7), and jenkins still do not try to rebuild.

I do not know what exactly problem you has with kinetic and qt5. Please add dump.
Next: i recommend to use `catkin` tool, not `catkin_make`.

`rosinstall_generator` simply pulls out package distribution data which `wstool` uses to retrieve sources.

siddhu45, 2016-07-28:
Thanks for the correction!
And the error which I get while building ROS Kinetic from source is while resolving the dependencies using the `rosdep` command: it shows `qtbase5-dev` failed to install.
The exact error I get when I run the command  `rosdep install --from-paths src --ignore-src --rosdistro kinetic -y' is
`E: Sub-process /usr/bin/dpkg returned an error code (1)ERROR: the following ros deps failed to install apt: command [sudo -H apt-get install -y qtbase5-dev] failed`

Thank you so much for your help!

vooon, 2016-07-28:
Why to build whole Kinetic distro? It takes a lot of time, event if you setup `distcc`.
Try to follow official instruction: http://wiki.ros.org/kinetic/Installation/Ubuntu

Also i don't thing that you actually need `desktop-full` on onboard computer, so try `base` or `robot`.

vooon, 2016-07-28:
Also check your APT mirrors, because official has http://packages.ubuntu.com/xenial/qtbase5-dev for xenial-armhf.

siddhu45, 2016-08-11:
Yea! followed the advice and now finally I'm able to install ROS Kinetic robot variant without any conflict of qt5 (had to change a few things though before installing to remove conflict). So after that I also installed MAVROS and it works fine.
Thank you so much for all your advises.

closed 2016-08-11 by siddhu45