mavlink/mavros #584: would it be possible to add support for cleanflight stack in mavros?

MohmadAyman, 2016-08-01:
from this PR, cleanflight now supports mavlink, how hard would it be to intergrate it with mavros?
https://github.com/cleanflight/cleanflight/pull/806
Thanks.

vooon, 2016-08-01:
In current state cleanflight support of mavlink is useless. They do not handle any message. You simply can't send any command to control drone.

To be usable they should support at least:
1. Settable system and maybe component ID. 0:200 - broadcast:200
2. Own AUTOPILOT_TYPE, not GENERIC - else you can't detect AP type
3. COMMAND_LONG and/or COMMAND_INT message handler. At least you should be able to ARM
4. Some types of setpoint messages. Else it useless for automatic control from mavros

Also good to implement parameter and mission protocols if possible.

After that on mavros side need to:
1. Add custom mode decode table
2. Make launch script and select what plugins is usable with that AP (pluginlists.yaml), add default config.yaml

And need someone to do integration tests, who has cleanflight hw. Also good if FW can be build in SITL mode.

TSC21, 2017-04-04:
Closing for now while there's no update on CleanFlight side. @MohmadAyman when you have an update on this let us know.

closed 2017-04-04 by TSC21