mavlink/mavros #588: APMcopter and Mavros

Muttle, 2016-08-11:
Hi,
 I have built a platform using an Intel Edison and Pixhawk which are mounted on a conventional helicopter using the APM flight stack. 

So far I have manage to get mavros to partially load but the work flow does not seem to be complete nor am I able to see topics such as imu data. 

What I would like to know is, have I hit a brick wall due to my flight stack(not being PX4) or is there some simple fixes that can be applied here to get full functionality of mavros/mavlink. 

I have included below is a copy of what initialised, warnings and errors.

started core service [/rosout]
process[mavros-2]: started with pid [3326]
[ INFO] [1470922018.771109791]: FCU URL: /dev/ttyMFD1:115200
[ INFO] [1470922018.773665725]: serial0: device: /dev/ttyMFD1 @ 115200 bps
[ INFO] [1470922018.789179274]: GCS bridge disabled
[ INFO] [1470922019.819122794]: Plugin 3dr_radio loaded and initialized
[ INFO] [1470922019.819774620]: Plugin actuator_control blacklisted
[ INFO] [1470922019.820079700]: Plugin altitude blacklisted
[ INFO] [1470922019.886691382]: Plugin command loaded and initialized
[ INFO] [1470922019.887133212]: Plugin ftp blacklisted
[ INFO] [1470922019.994099769]: Plugin global_position loaded and initialized
[ INFO] [1470922020.003513203]: Plugin hil_controls loaded and initialized
[ INFO] [1470922020.104156898]: Plugin imu_pub loaded and initialized
[ INFO] [1470922020.184243786]: Plugin local_position loaded and initialized
[ INFO] [1470922020.193549297]: Plugin manual_control loaded and initialized
[ INFO] [1470922020.227403564]: Plugin param loaded and initialized
[ INFO] [1470922020.282628444]: Plugin rc_io loaded and initialized
[ INFO] [1470922020.283719496]: Plugin safety_area blacklisted
[ INFO] [1470922020.336800864]: Plugin setpoint_accel loaded and initialized
[ INFO] [1470922020.337229913]: Plugin setpoint_attitude blacklisted
[ INFO] [1470922020.428873196]: Plugin setpoint_position loaded and initialized
[ INFO] [1470922020.565545018]: Plugin setpoint_raw loaded and initialized
[ INFO] [1470922020.604502269]: Plugin setpoint_velocity loaded and initialized
[ INFO] [1470922020.688886286]: Plugin sys_status loaded and initialized
[ INFO] [1470922020.746002092]: Plugin sys_time loaded and initialized
[ INFO] [1470922020.763078851]: Plugin vfr_hud loaded and initialized
[ INFO] [1470922020.816650847]: Plugin waypoint loaded and initialized
[ INFO] [1470922020.817153674]: Built-in SIMD instructions: None
[ INFO] [1470922020.817455428]: Built-in MAVLink package version: 2016.5.20
[ INFO] [1470922020.817722187]: Built-in MAVLink dialect: ardupilotmega
[ INFO] [1470922020.818022068]: MAVROS started. MY ID 1.240, TARGET ID 1.1
-_-
[ INFO] [1470922021.263817973]: CON: Got HEARTBEAT, connected. FCU: ArduPilotMeg                                       a
[ WARN] [1470922023.276634796]: VER: broadcast request timeout, retries left 4
[ERROR] [1470922024.268819229]: FCU: PreArm: Need 3D Fix
[ WARN] [1470922024.277118735]: VER: broadcast request timeout, retries left 3
[ WARN] [1470922030.276962185]: VER: unicast request timeout, retries left 2
[ WARN] [1470922032.266740877]: PR: request list timeout, retries left 2
[ WARN] [1470922033.267823928]: PR: request list timeout, retries left 1
[ WARN] [1470922034.268991856]: PR: request list timeout, retries left 0
[ WARN] [1470922035.290013330]: VER: unicast request timeout, retries left 1
[ WARN] [1470922037.266418279]: WP: timeout, retries left 2
[ WARN] [1470922038.268264135]: WP: timeout, retries left 1
[ WARN] [1470922039.270417925]: WP: timeout, retries left 0
[ERROR] [1470922040.271798814]: WP: timed out.
[ WARN] [1470922040.302990345]: VER: unicast request timeout, retries left 0
[ WARN] [1470922045.315961221]: VER: your FCU don't support AUTOPILOT_VERSION, s                                       witched to default capabilities

Thanks

Tom

vooon, 2016-08-11:
Run `rosrun mavros checkid` with running mavros launch. Use ``` for code blocks.

Muttle, 2016-08-12:
Hi vooon,

This is my result

OK. I got messages from 1:1.

---
Received 15 messages, from 1 addresses
sys:comp   list of messages
  1:1     0

Was unsure how to implement the ``` for code blocks.

Thanks

vooon, 2016-08-12:
https://help.github.com/articles/creating-and-highlighting-code-blocks/

Hmm, FCU only sends HEARTBEAT messages, no response on parameter request and mission list request.
But since HEARTBEAT present there no baudrate error.

Check electrical connection of Edison Tx ←→ Pixhawk Rx and GND ←→ GND.
Then use GCS to check TELEM2 connection params (`SERIAL%d_xxx`), for now better to use MAVLink 1.

Muttle, 2016-08-14:
Thanks! I had the tx on the wrong pin on the edison but I still get this

Received 15 messages, from 1 addresses
sys:comp   list of messages
  1:1     0

However now i get this 
```
[ INFO] [1471166540.155757162]: CON: Got HEARTBEAT, connected. FCU: ArduPilotMeg                       a
[ INFO] [1471166541.185158331]: VER: 1.1: Capabilities 0x0000000000000b83
[ INFO] [1471166541.185876875]: VER: 1.1: Flight software:     030303ff (acf2e10                       c34e1d5437c5ef883???
??      )
[ INFO] [1471166541.186887307]: VER: 1.1: Middleware software: 00000000 (34e1d54                       37c5ef883???
??      )
[ INFO] [1471166541.187820265]: VER: 1.1: OS software:         00000000 (7c5ef88                       3???
??      )
[ INFO] [1471166541.188606918]: VER: 1.1: Board hardware:      00000000
[ INFO] [1471166541.189224991]: VER: 1.1: VID/PID: 0000:0000
[ INFO] [1471166541.189839447]: VER: 1.1: UID: 0000000000000000
[ WARN] [1471166541.190596102]: CMD: Unexpected command 520, result 0
PuTTYPuTTYPuTTY[ERROR] [1471166550.201329733]: FCU: APM:Copter V3.3.3 (acf2e10c)
[ERROR] [1471166550.202260968]: FCU: PX4: 34e1d543 NuttX: 7c5ef883
[ERROR] [1471166550.202956265]: FCU: Frame: HELI
[ERROR] [1471166550.204280701]: FCU: PX4v2 00460028 3135510C 35333436
[ INFO] [1471166555.165624514]: WP: mission received
[ERROR] [1471166558.161771914]: FCU: PreArm: Need 3D Fix
[ INFO] [1471166561.380437222]: PR: parameters list received
[ERROR] [1471166588.160883805]: FCU: PreArm: Need 3D Fix
[ERROR] [1471166618.159717800]: FCU: PreArm: Need 3D Fix
[ERROR] [1471166648.166872454]: FCU: PreArm: Need 3D Fix
```

vooon, 2016-08-14:
Check diagnostics, then use `rqt` to check message rates.

Muttle, 2016-08-16:
Hi Vooon, 

I have used rostopic hz instead to check if topics such as imu/data are publishing, which they are not. The only topic which seems to work is /mavros/state.

Diagnostics seems ok and gives more data when the drone is armed.

Can you please advise further? Do I still need to run rqt to check the message rates? 

Thanks

Muttle, 2016-08-16:
Actually.. Got it to work by changing the message stream rate.

rosservice call /mavros/set_stream_rate 0 10 1

Except my quaternions dont seem to publish in my imu data..

closed 2016-08-18 by Muttle