mavlink/mavros #589: Increase the data stream rate in px4 stack using mavros

hridaybavle, 2016-08-12:
Hi, 
I want to increase the publish rate of the sensors given by pixhawk. I tried many things 

1. Increasing the baud rate - does not work 
2. [Increase output frequency of pixhawk using mavros](http://answers.ros.org/question/213390/increase-output-frequency-of-pixhawk-using-mavros/)
3. mavros service (set_stream_rate)

I am still not able to increase the frequency of any sensor. Is there any way to increase the frequency that I am not taking into account. 

PS: I want to increase the frequency of lidarlite connected to pixhawk serial port (it publishes at only 10 hz), is it possible to do it using mavros ?
Thanks.

vooon, 2016-08-12:
Did you configure SYS_COMPANION for companion link?

Did you sure that rangefinder has poll rate > 10 Hz?

Did you configure stream rate via `extras.txt`?

mhkabir, 2016-08-12:
@vooon PX4 now supports per-message stream rate setting. Maybe we could add this feature in mavros now?

hridaybavle, 2016-08-12:
1. Yes I put in pixhawk SYS_COMPANION = Companion Link (921600 baud)
2. I added 
mavlink stream -d /dev/ttyACM0 -s ATTITUDE -r 200
mavlink stream -d /dev/ttyACM0 -s HIGHRES_IMU -r 200 in etc/extras.txt

The rate of the IMU increased to 160 Hz but I see a rare thing, that in order to get the IMU publishing at 160Hz I first have to connect the pixhawk to a the companion computer and then power power the pixhawk using battery. 
But if I power the pixhawk first with a battery and then connect it to the companion computer the IMU data is published at 80 Hz, I don't know it its a normal behaviour. 

**3. Did you sure that rangefinder has poll rate > 10 Hz?**
I have a laser range finder SF/10A which I connect using this [link](https://pixhawk.org/peripherals/rangefinder) 
I am not sure how to check the poll rate of the range finder, but I still get the frequency between 9-10Hz.

vooon, 2016-08-12:
@mhkabir that was #370. Did you have bandwidth for that?

vooon, 2016-08-12:
`ttyACM0` is not companion link! You should use TELEM2 port & UART adapter for that.

Old, but i hope still working method:
https://github.com/vooon/rarog/blob/master/rarog_configs/pixhawk/extras.txt

vooon, 2016-08-12:
As i understand new logger (.ulg) write uORB topics at it's maximum rate, so perhaps SD logs may help to investigate real poll rate.
Also good point to read device driver source. It relatively simple.

hridaybavle, 2016-09-19:
At the end I was able to increase the speed of the sensors by adding in etc/extras.txt in the SD 
mavlink stream -d /dev/ttyACM0 -s HIGHRES_IMU -r 200 

But I was able to increase the speed of the Lidar sensor only from 10hz to 12hz, I guess its the lidar driver in the pixhawk, which does not publish more than 12 hz. 

Thanks.

closed 2016-09-19 by hridaybavle

mhkabir, 2016-09-19:
> But I was able to increase the speed of the Lidar sensor only from 10hz to 12hz, I guess its the lidar driver in the pixhawk, which does not publish more than 12 hz.

Limitations of running it over PWM.

SubMishMar, 2018-03-28:
Is it possible to do this over the TELEM2 port? I want the IMU to be running at more than 100 Hz.

Jdiesel666, 2018-05-14:
@hridaybavle 
Hi I just started to use pixHawk and ran into the problem of not being able to increase the frequency. Could you help walk me through the process? Also, I found it weird that I don't have that extras.txt file you were talking about. Do I need to create it by myself inside some folder?

Mu99-M, 2023-05-19:
Have you been able to find a solution for it?
I'm not able to increase the frequency of `VEHICLE_ANGULAR_VELOCITY` to more than 230 Hz in Mavros, however, it's being published at 800 Hz when I run `uorb top` in the Mavlink shell.