mavlink/mavros #590: Rosbag records empty file

Muttle, 2016-08-18:
Hi all,

I am trying to use rosbag to record certain topics on my drone for an experiment. I am using an edison coupled with the px4 on a conventional helicopter running APM arducopter 3.3.3.

Now, due to a lack of free space on my edison, I am forced to used sshfs to mount an external drive and store the file offboard. I understand that this is not desirable however, I do not need extremely high sample rates and will only be concerned with a few topics so this should be ok? 

My problem is when I record topics no data appears in the file when I play it back or whilst it is running.

Thanks in advance.

vooon, 2016-08-18:
What `rqt_bag` shows?

Muttle, 2016-08-18:
Hi vooon, 

rqt_bag shows it is all there. 

For some reason (probably me) I could not plot it using rqt_plot. But I have had a friend plot it on another machine.

Thanks!

closed 2016-11-01 by vooon