mavlink/mavros #591: ALTCTL & POSCTL in MOCAP with Pixhawk + Raspberry Pi + MAVROS on quadcopter in MOCAP [optitrack]

siddhu45, 2016-08-19:
Hello community,
I'm trying to do Altitude & Position control on my quad with Pixhawk and Raspberry Pi onboard. I send commands through MAVROS to the onboard computer via MAVROS from the ground station. I performed certain experiments and so far I was not able to achieve any of the aforementioned flight modes when flying in the MOCAP environment. I have had certain issues and questions:

1. I transformed the coordinate frame of the MOCAP data I'm getting from the Motive software (and which I'm forwarding to my uav) but still I have about 80 deg of offset in yaw and about 2 deg in roll. How is it? Can anyone help me to properly align the coordinate frames of my uav and mocap system. As of my knowledge, by default, the mocap follows the right handed coordinate reference while the Pixhawk follows NED. Is it correct?

2. I'm flying without GPS in the indoor environment and have set all GPS weights to 0. After setting everything for mocap, I could not achieve the desired results so I mapped everything to vision, as suggested in some of the posts, but still no luck. Altitude control worked till certain extent but not for long. The uav starts to deviate more and more. Same goes for the Position control, the moment I put my quad to Position control mode, it starts drifting and circling aggressively. Why so? I check the data I am getting from mocap and the local data from Pixhawk are the same for position. The orientation quaternions do not match with each other as I'm guessing I haven't solved the transformation issue completely. But atleast it should follow the position correctly.

3. I have 2 different packages, mocap_optitrack and optitrack (newer), but I'm not able to use the newer package for my experiments. Ok, maybe this is not an urgent Q but still any help would be great.

I've been working around uavs for while quite a while but just started working with MOCAP system, so any help to get me up and running would be really appreciated. I could give more data if you need but my lab is under renovation will next week. So maybe a few instructions or tips or anything relevant would be great!

Thanks a lot guys

siddhu45, 2016-08-22:
Is my question clear or more description is required?
I am really stuck and have read and tried to understand from other issues but not able to find my answers. So, if anyone could help me please!

Thanks

vooon, 2016-08-22:
Try to ask help http://discuss.px4.io/ (PX4 support forum).

I'm not used any mocap system (and copters too), so i can't help. But 80 deg offset looks weird. Frame rotation errors should look as multiplication of 90 deg. What difference with this packages (link?), and why new one is not usable?

siddhu45, 2016-08-25:
Ok. Thanks for the suggestion, I'll try that too.
And exactly, 80 deg offset is weird for me too. About the new package! I am not able to receive any data and I couldn't check as the codes are written in python (and I not good with python, I can read them if it was in C/C++).
Anyways, Thanks for your help

closed 2016-09-13 by siddhu45

mechy45, 2016-09-24:
Hi siddhu45,

My case is similar to yours.

How did you solve it?

siddhu45, 2016-09-28:
Hey @mechy45 
I did the same things as mentioned about the transformation as it is the key to solving this issue. What I learned s you need to define a rigid body in MOTIVE in a way that the local x-axis and the x-axis of your MOCAP system are aligned. The MOCAP has North-West_Up and the Pixhawk on my quad follows East-West-Down. Other main things you need to take care of  are the sensor weights. What controller are you using, Pixhawk? Let me know what exactly are you looking for! I'll help if I can

Regards
Sid

nahush8, 2017-02-23:
@siddhu45  I am facing the exact same issue. Mocap and local position data are exact same but as soon as I switch to position control,or altitude control it starts to drift.  Can you please help me with the steps of fixing this problem ?

siddhu45, 2017-02-23:
Are you using PIXHAWK? If so, then you need to confirm that the data from mocap is correctly being published through MAVROS. Check if you have assigned the weights properly. There are 2 ways to do that: Vision and Mocap. I tried both and both worked for me. Increase these weights to almost maximum and minimize the GPS weights as we are flying indoors. Other than that I would say make sure the mocap and local data are in sync. 
How much is the drift? Is it sudden and agile or slow and steady? I hope the above mentioned things help you solve the issue.
Cheers!

nahush8, 2017-02-23:
@siddhu45  
1. Yes, I am using pixhawk. I am using the lpe firmware which has the lpe_fusion parameter. (I only select vision there.) 
2. I am using mocap : /mavros/mocap/pose and /mavros/local_position/pose are in sync for x,y,z but orientation values don't match. (I checked in RVIZ and the axes for two topics are aligned.)
3. Can you tell me which weight parameters should i check in pixhawk ?
4. The drift is kind of sudden and quick. As soon as I switch to POSCTL or ATLCTL from manual, i see a lot of drift especially in XY direction. Altitude seems to be a bit stable.

siddhu45, 2017-02-23:
1. Don't use lpe, instead use INAV. Which PIXHAWK firmware version are you using? Do not use 1.5+ right now, as there are some conflicts with the mocap parameters in that. Try using 1.4.4
2. The orientation these topic show is in quaternions, convert them to Euler angles and they should match. How are you publishing the data from mocap? A simple way to match (how I did it) is, define a rigid body, connect and try to compare the local pose and the mocap values in the software. Then once you match the axes then you will see that they might be inverted. So, in your publisher script you have to change accordingly, x-y-z and roll-pitch-yaw. (This is very important, I crashed 2 quads because of this mismatch)
3. Yes, I am talking about the ATT_W_ACC, ATT_W_EXT_HDG (mocap and vision, when you use INAV).
4. If the drift still persists, decrease the barometric and magnetic sensor weights, INAV_W_Z_BARO etc.

nahush8, 2017-02-23:
@siddhu45  I am using the latest release and uploading the px4fmu-v2_lpe firmware.  
1. From mocap, I am publishing on a topic /mavros/mocap/pose which is picked up by the mocap_pose_estimate plugin of mavros_extras ,( I am assuming this step is correct because if it's not , then i have a fundamental problem to solve). After this, my /mavros/local_position/pose and /mavros/mocap/pose match perfectly for x, y and z. (This is Not orientation values, but pose published, so does this mean that they both match well and i don't have to do anything in that ? )
2. okay i will check if varying these parameters in the older firmware make a difference. (When I did it with lpe firmware, it just crashed a multiple times.)

Thanks a lot for all your inputs. I will test more and get back with results

sahabi, 2017-04-12:
Hey @nahush8,

Did it work out for you? Were the suggested fixes by @siddhu45 sufficient for solving your issue?

We are having the exact same issue with LPE. I also noticed that the velocity estimates are off after an aggressive episode when switching to OFFBOARD position control.

Thanks!

AnkithThodupunoori, 2021-12-08:
@SiddharthPatel45 , i am trying do the same thing which you have done in your project, i am facing some trouble in figuring out how to communicate between mocap and raspberry pi, could you please give me some instructions about the procedure how to do it, we can communicate trough email if you would like to.

SiddharthPatel45, 2021-12-09:
You might need the [mocap_optitrack](https://wiki.ros.org/mocap_optitrack) package to do that. Install it on your Raspberry Pi and configure the mocap software (eg: motive for OptiTrack) to broadcast data on your network. If you are using another source than you might need to bridge that using custom SDK and publish on the mavros's mocap topics for your robot/drone to receive. Hope this helps!

vipin3690, 2022-10-24:
hi @SiddharthPatel45 , have the same issue like same data on position but orientation is different.I am using pixhawk cube orange  and raspi.The details are available on https://github.com/mavlink/mavros/issues/1792