mavlink/mavros #593: MAVROS not detecting PX4's AUTOPILOT_VERSION replies (MAVLink 2 related?)

LorenzMeier, 2016-08-27:
@vooon I'm getting this error message:
https://github.com/mavlink/mavros/blob/6c36688c38bd81fb8ce859938e77e2c5bfa276b6/mavros/src/plugins/sys_status.cpp#L848

In this test:
http://sitl01.dronetest.io/job/PX4%20Firmware%20POSIX%20SITL%20PRs/2036/console

What might be notable is that the software in the test supports MAVLink 2.0. Does the error indicate that the communication is just not working?

vooon, 2016-08-27:
Looks like no response to `COMMAND_LONG`. Perhaps firmware drops message?

Note that you use version without 2.0 support. I guess 0.17?

LorenzMeier, 2016-08-27:
I could repro this since without a MAVLink change. Sorry for the noise.

closed 2016-08-27 by LorenzMeier