mavlink/mavros #594: Mav ignores "coordinate_frame" when using mavros_msgs/PositionTarget

Pryre, 2016-08-29:
As a precursor, I can't figure out if this is a firmware side issue, or a mavros issue.

I have some code that is performing velocity control in the body frame of the mav, but I can only seem to get it to move with respect to the local frame. I am setting the coordinate frame as "FRAME_BODY_OFFSET_NED", and applying the type_mask as appropriate, but no matter what, the mav only moves in respect to the world NED, and not body frame NED.

The code is relatively basic, and doesn't perform rotations to account for yaw in the world frame. If I change "coordinate_frame" to any of the available options, the mav always flies in the "correct" path regardless of yaw. Basically I cannot command the UAV to fly forward with respect to its own heading.

Does anyone know what or where this issue is being caused?

vooon, 2016-08-29:
As you can see in [setpoint_raw.cpp L163][1], it's firmware side.

[1]: https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/setpoint_raw.cpp#L163

Pryre, 2016-08-29:
Thanks, was mainly after confirmation.

closed 2016-08-29 by Pryre