mavlink/mavros #598: question: handle_local_position_ned(), the pose.x is puzzled me...

asimay, 2016-09-12:
In handle_local_position_ned() function, the odom->pose.pose has x & y elements. so the odom->pose.pose.x is equal to Vx *deltT, or ve * deltT? (Vx is of body-enu frame, ve is local enu frame). if sensor didn't report x, y, z directly, I don't know use which velocity to calculate the pose?

vooon, 2016-09-12:
Please give me a link to source file + line (click on line number, copy & paste url).

asimay, 2016-09-12:
Thank you very much ! @vooon .
https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/local_position.cpp#L116

asimay, 2016-09-13:
I mean: if we calculate the pose according to the velocity, here is some ambiguous. should 

> pose.x = vx * deltT 
or

> pose.x = ve * deltT 
? this confuse me a lot. or use transform_frame_baselink_enu() to transform body's position to enu position, but this still need to calculate body position.

vooon, 2016-09-13:
What is *vx*, *ve*? Define please. On L116 pose message copied to odom. `pose.position.{xyz} = enu_position.{xyz}`

asimay, 2016-09-14:
sorry for misleading. 
vx is vehicle-body velocity x WRT ENU. ve is east velocity of INS output WRT NED.

vooon, 2016-09-14:
Position provided by FCU. Only frame rotation needed.

closed 2017-01-23 by vooon