mavlink/mavros #599: Hover and POSCTL of UAV with Pixhawk + Raspberry Pi onborad using MOCAP Real-time feedback

siddhu45, 2016-09-13:
Hello community,
Previously I was trying to achieve this feedback but had problem with the position and orientation matching. But now I have solved that issue so I am able to give very accurate position & orientation commands based on the mocap data to the Pixhawk!
Although the issues I am facing are:
1. I am not able to hover whenever I change the flight mode to POSCTL. I also tried commanding through ROS via Raspberry Pi to hover but it still doesn't do that. It will hover for a few seconds and then start to drift and perform circular motions aggressively. Why is this?
2. Also, what should be range of weights given to other sensors like barometer, GPS [indoor environment], IMU, magnetometer etc. Will it be helpful to map every thing to vision instead of mocap in parameters in QGC.

All and any help is really appreciated.
Thanks guys

siddhu45, 2016-09-13:
@jespestana @BrownKyle
I read on other issue that you guys have already achieved what I'am trying to do. I'll be grateful if you guys can help me out!
Thank you so much

JGSOpenSrc, 2016-09-27:
Hi there Siddhu!

I'm working on the same things as you are but I don't think I am as far along. I am having trouble getting the mocap plugin of mavros to work.

But that is besides the point. My guess is that your problem is more likely an issue with your flight stack, rather than with mavros. Are you using PX4, or Ardupilot? If you are using PX4, and you haven't already, head over to the [px4 discussion page](discuss.px4.io) and the [developer documentation](dev.px4.io).

I wish you the best of luck with your project!

siddhu45, 2016-09-28:
Hey @JGSOpenSrc, Thanks for the responce!
Yea I'm using PX4 Flight stack. But just to inform you, I solved my issues and now I can do POSCTL & ALTCTL on my uav and also can run a simple script to hover my quad in OFFBOARD. Which ROS plugin are you using? As far as I know, there are 2 of them, "mocap_optitrack" and "optitrack". the latter is newer and I couldn't make it work (still trying on that) but the former package I can use. Let me know if you have some other doubts regarding this on the other issue of mine:
[](https://github.com/mavlink/mavros/issues/591#issuecomment-249349335)

Thanks
Sid

closed 2016-09-28 by siddhu45