mavlink/mavros #600: Issue with 3DR Solo drone: "PR: request param #41 timeout, ..."

micheda, 2016-09-16:
Hi,

I'm trying to control my 3DR Solo drone with MAVROS:

roslaunch mavros px4.launch fcu_url:="udp://:14550@10.1.1.10:14555"

I get tons of these warnings:

"[ WARN] [1473858171.706365027]: PR: request param #41 timeout, retries left 2, and 331 params still missing"

I can receive all parameters using qgroundcontrol , so this is probably something to fix in MAVROS.

Anyone that can help us understand what's wrong?

Thanks
Michele

vooon, 2016-09-16:
Try `checkid` script.

toolazi, 2016-11-22:
Hi I have the same problem as micheda did above. I ran the checkid script and got:

OK. I got messages from 1:1.

---
Received 698 messages, from 3 addresses
sys:comp   list of messages
  1:154   200, 215
 10:0     109
  1:1     33, 35, 36, 129, 178, 116, 22, 27, 29, 30

I am new to this and do not know how to interpret this or how to use it to fix the problem. Any help would be great.

Thanks!

vooon, 2016-11-23:
@toolazi versions of fw? mavros? Log output (console where you run apm.launch).

toolazi, 2016-11-29:
@vooon I am not exactly sure what version of the firmware I am using. It is the stock fw on the 3dr Solo and for mavros I believe it to be v0.18.0. To install mavros is followed http://dev.px4.io/ros-mavros-installation.html. The log output (after I run apm.launch) looks like this:

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 0.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 0.0
 * /mavros/fcu_url: udp://:14550@10.1...
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: fcu
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: True
 * /mavros/mocap/use_tf: False
 * /mavros/plugin_blacklist: ['actuator_contro...
 * /mavros/plugin_whitelist: []
 * /mavros/px4flow/frame_id: px4flow
 * /mavros/px4flow/ranger_fov: 0.0
 * /mavros/px4flow/ranger_max_range: 5.0
 * /mavros/px4flow/ranger_min_range: 0.3
 * /mavros/safety_area/p1/x: 1.0
 * /mavros/safety_area/p1/y: 1.0
 * /mavros/safety_area/p1/z: 1.0
 * /mavros/safety_area/p2/x: -1.0
 * /mavros/safety_area/p2/y: -1.0
 * /mavros/safety_area/p2/z: -1.0
 * /mavros/setpoint_accel/send_force: False
 * /mavros/setpoint_attitude/reverse_throttle: False
 * /mavros/setpoint_attitude/tf/child_frame_id: attitude
 * /mavros/setpoint_attitude/tf/frame_id: local_origin
 * /mavros/setpoint_attitude/tf/listen: False
 * /mavros/setpoint_attitude/tf/rate_limit: 10.0
 * /mavros/setpoint_position/tf/child_frame_id: setpoint
 * /mavros/setpoint_position/tf/frame_id: local_origin
 * /mavros/setpoint_position/tf/listen: False
 * /mavros/setpoint_position/tf/rate_limit: 50.0
 * /mavros/startup_px4_usb_quirk: False
 * /mavros/sys/disable_diag: False
 * /mavros/sys/min_voltage: 10.0
 * /mavros/target_component_id: 1
 * /mavros/target_system_id: 1
 * /mavros/tdr_radio/low_rssi: 40
 * /mavros/time/time_ref_source: fcu
 * /mavros/time/timesync_avg_alpha: 0.6
 * /mavros/vibration/frame_id: vibration
 * /mavros/vision_pose/tf/child_frame_id: vision
 * /mavros/vision_pose/tf/frame_id: local_origin
 * /mavros/vision_pose/tf/listen: False
 * /mavros/vision_pose/tf/rate_limit: 10.0
 * /mavros/vision_speed/listen_twist: False
[... 50 lines cut ...]

and immediately after looks like this:

[ WARN] [1480443647.777340422]: TM: Wrong FCU time.
[ WARN] [1480443648.763964962]: IMU: linear acceleration on RAW_IMU known on APM only.
[ WARN] [1480443648.764185235]: IMU: ~imu/data_raw stores unscaled raw acceleration report.
[ INFO] [1480443654.299410741]: CON: Got HEARTBEAT, connected. FCU: ArduPilotMega
[ INFO] [1480443655.473641513]: VER: 1.1: Capabilities 0x0000000000000000
[ INFO] [1480443655.474057066]: VER: 1.1: Flight software:     00000000 (7e9206cc5e693274d48fa307?)
[ INFO] [1480443655.474364997]: VER: 1.1: Middleware software: 00000000 (5e693274d48fa307?)
[ INFO] [1480443655.474927264]: VER: 1.1: OS software:         00000000 (d48fa307?)
[ INFO] [1480443655.475082379]: VER: 1.1: Board hardware:      00000000
[ INFO] [1480443655.475232640]: VER: 1.1: VID/PID: 0000:0000
[ INFO] [1480443655.475363855]: VER: 1.1: UID: 0000000000000000
 
which is followed by:
[ERROR] [1480443657.291096294]: FCU: PreArm: Need 3D Fix
[ WARN] [1480443661.140008691]: RADIO_STATUS not from 3DR modem?
[ERROR] [1480443665.427236388]: FCU: PreArm: Need 3D Fix
[ WARN] [1480443667.653980115]: PR: request param #0 timeout, retries left 2, and 426 params still missing
[ WARN] [1480443668.752541494]: PR: request param #1 timeout, retries left 2, and 425 params still missing
[ WARN] [1480443669.852170120]: PR: request param #2 timeout, retries left 2, and 424 params still missing
[ WARN] [1480443670.302396676]: WP: timeout, retries left 2

Thanks.

vooon, 2016-11-29:
0.18 has serious issues, update to latest. But i'm not sure that it helps.
Try to call some commands. E.g.
```
rosrun mavros mavcmd long -b 520 1 0 0 0 0 0 0
```

Check telemetry upstream path (ground to drone).

toolazi, 2016-12-07:
Hi @vooon. Turns out my mavros was at v0.17.4. That is the version that is automatically installed when following http://dev.px4.io/ros-mavros-installation.html. I know it might be very trivial but how can I update mavros to the latest version? Again, I am new to a lot of this so I appreciate your patience and help!

Thanks.

vooon, 2016-12-07:
What method selected? It is binary installation? If so, then apt remove them.

Then follow  https://github.com/mavlink/mavros/blob/master/mavros/README.md#source-installation
NOTE: both rosinstall_generator should be called with `--rosdistro=kinetic`.

But if you could, better to upgrade to Kinetic (require Ubuntu 16.04 / Debian).

closed 2017-01-23 by vooon

kandai-wata, 2017-11-12:
Is this problem fixed? if so, could you tell me how? I got the same problem with 3DR Solo **indoor**.  
I ran these commands (pasted below!) 
``` 
roslaunch mavros px4.launch fcu_url:=udp://:14550@10.1.1.10:14555
rosrun mavros checkid
rostopic echo /diagnostics
```
The reasons I could think of was, 
1. GPS signal not received due to the indoor environment, so it can not finish initialization.
2. Flight Stack is Ardupilot, so it cannot handle some mavros topics like /mavros/setpoint_attitude (QGroundControl should be able to set waypoints outdoor so this may not be a problem) 

Ros: Indigo
Mavros: 0.20.0 (I'm not sure how to check)
Ubuntu: 14.04 LTS
FCU: Pixhaw 2.0
Firmware: PX4Firmware-solo
Flight Stack: Ardupilot (as default for 3DR Solo)

This is a bit off topic but, if I want to send mocap position data to Solo for further development, do I need to upload PX4 firmware, or is it possible with APM?
If PX4 is needed, should I upload PX4Firmware master or PX4Firmware-solo?

```
watanabe@ytnpc2016a-W230SD ~ $roslaunch mavros apm.launch fcu_url:=udp://:14550@10.1.1.10:14555
... logging to /home/watanabe/.ros/log/1524b312-c74b-11e7-96fa-80fa5b2803ee/roslaunch-ytnpc2016a-W230SD-28509.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ytnpc2016a-W230SD:59061/

SUMMARY
========

CLEAR PARAMETERS
* /mavros/

PARAMETERS
* /mavros/cmd/use_comp_id_system_control: False
* /mavros/conn/heartbeat_rate: 1.0
* /mavros/conn/system_time_rate: 0.0
* /mavros/conn/timeout: 10.0
* /mavros/conn/timesync_rate: 0.0
* /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
* /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
* /mavros/distance_sensor/rangefinder_pub/id: 0
* /mavros/distance_sensor/rangefinder_pub/send_tf: False
* /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
* /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
* /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
* /mavros/distance_sensor/rangefinder_sub/id: 1
* /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
* /mavros/distance_sensor/rangefinder_sub/subscriber: True
* /mavros/fake_gps/eph: 2.0
* /mavros/fake_gps/epv: 2.0
* /mavros/fake_gps/fix_type: 3
* /mavros/fake_gps/geo_origin/alt: 408.0
* /mavros/fake_gps/geo_origin/lat: 47.3667
* /mavros/fake_gps/geo_origin/lon: 8.55
* /mavros/fake_gps/gps_rate: 5.0
* /mavros/fake_gps/mocap_transform: True
* /mavros/fake_gps/satellites_visible: 5
* /mavros/fake_gps/tf/child_frame_id: fix
* /mavros/fake_gps/tf/frame_id: map
* /mavros/fake_gps/tf/listen: False
* /mavros/fake_gps/tf/rate_limit: 10.0
* /mavros/fake_gps/tf/send: False
* /mavros/fake_gps/use_mocap: True
* /mavros/fake_gps/use_vision: False
* /mavros/fcu_url: udp://:14550@10.1...
* /mavros/gcs_url:
[... 444 lines cut ...]
key: RSSI (dBm)
value: -127.0
-
key: Remote RSSI
value: 231
-
key: Remote RSSI (dBm)
value: -5.4
-
key: Tx buffer (%)
value: 0
-
key: Noice level
value: 0
-
key: Remote noice level
value: 0
-
key: Rx errors
value: 7
-
key: Fixed
value: 0
---
```

alexwb1, 2018-04-06:
I am having this problem too and I need help!!