mavlink/mavros #601: Mavros request fail issue

yunshin, 2016-09-18:
Hello.
I'm trying to connect pixhawk to Nvidia jetson TX1 board.
I installed ROS Indigo including mavros on TX1.
However, everytime I try to use command:$ mavros mavsys mode -c OFFBOARD, it says mavros request fail.
When I type $dmesg

[image]

So I set fcu_url value to /dev/ttyACM0:57600 in my px4.launch file as below.

[image]

After then, I run roscore and roslaunch mavros px4.launch

`ubuntu@tegra-ubuntu:~/ros_catkin_ws/install_isolated/share/mavros/launch$ roslaunch mavros px4.launch
... logging to /home/ubuntu/.ros/log/896a075a-7d95-11e6-9e8c-00044b5809b8/roslaunch-tegra-ubuntu-1295.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.0.21:33842/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_url: /dev/ttyACM0:57600
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: fcu
[... 245 lines cut ...]
        value: 0
      - 
        key: Frequency (Hz)
        value: 0.000000
      - 
        key: Last dt (ms)
        value: 0.000000
      - 
        key: Mean dt (ms)
        value: 0.000000
      - 
        key: Last system time (s)
        value: 0.000000000
      - 
        key: Time offset (s)
        value: 0.000000000
---

`
 In Level 1, it says 'not connected'.  
How can I fix this connection problem??  
I tried to set baudrate as 57600, 921600 and 1152000 but it did not work.

+ I checked out that the connection between pixhawk and my desktop is fine. Only the connection between pixhawk and TX1 is the problem. 
Any advice will be very thankful to me.

closed 2016-09-18 by yunshin

KKLaLaLa, 2019-04-24:
hi ,i have meet the same problem with you.I solve it though modify px4.launch file. 
first you should us ls -a /dev to look which serial you use. ttyusb0 or ttyacm0.when i use ttyusb0.It have the same problem like you.if you use ttyacm0.then ecernthong will be ok.

chaqing, 2020-10-04:
Can you elaborate more?Thanks!

chaqing, 2020-10-04:
I meet the same problemwith you.Can you elaborate more?Thanks!