mavlink/mavros #602: Mavros request fail issue

yunshin, 2016-09-18:
Hello.
I'm trying to connect pixhawk to Nvidia jetson TX1 board.
I installed ROS Indigo including mavros on TX1.
However, everytime I try to use command:$ mavros mavsys mode -c OFFBOARD, it says mavros request fail.
When I type $dmesg

`[ 8812.276651] usb 1-3.4: new full-speed USB device number 14 using tegra-xhci
[ 8812.301674] usb 1-3.4: string descriptor 0 malformed (err = 2), defaulting to 0x0409
[ 8812.312785] usb 1-3.4: New USB device found, idVendor=26ac, idProduct=0011
[ 8812.312910] usb 1-3.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 8812.313012] usb 1-3.4: Product: PX4 BL FMU v2.x
[ 8812.313103] usb 1-3.4: Manufacturer: 3D Robotics
[ 8812.313189] usb 1-3.4: SerialNumber: 0
[ 8812.317142] usb 1-3.4: ep 0x83 - rounding interval to 1024 microframes, ep desc says 2040 microframes
[ 8812.320313] cdc_acm 1-3.4:1.0: This device cannot do calls on its own. It is not a modem.
[ 8812.330279] cdc_acm 1-3.4:1.0: ttyACM0: USB ACM device`

So I set fcu_url value to /dev/ttyACM0:57600 in my px4.launch file as below.

`<launch>
        
        

        <arg name="fcu_url" default="/dev/ttyACM0:57600" />
        <arg name="gcs_url" default="" />
        <arg name="tgt_system" default="1" />
        <arg name="tgt_component" default="1" />
        <arg name="log_output" default="screen" />

        <include file="$(find mavros)/launch/node.launch">
                <arg name="pluginlists_yaml" value="$(find mavros)/launch/px4_pluginlists.yaml" />
                <arg name="config_yaml" value="$(find mavros)/launch/px4_config.yaml" />

                <arg name="fcu_url" value="$(arg fcu_url)" />
                <arg name="gcs_url" value="$(arg gcs_url)" />
                <arg name="tgt_system" value="$(arg tgt_system)" />
                <arg name="tgt_component" value="$(arg tgt_component)" />
                <arg name="log_output" value="$(arg log_output)" />
        </include>
</launch>
`

After then, I run roscore and roslaunch mavros px4.launch

`ubuntu@tegra-ubuntu:~/ros_catkin_ws/install_isolated/share/mavros/launch$ roslaunch mavros px4.launch
... logging to /home/ubuntu/.ros/log/896a075a-7d95-11e6-9e8c-00044b5809b8/roslaunch-tegra-ubuntu-1295.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.0.21:33842/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
[... 275 lines cut ...]
        value: 0
      - 
        key: Frequency (Hz)
        value: 0.000000
      - 
        key: Last dt (ms)
        value: 0.000000
      - 
        key: Mean dt (ms)
        value: 0.000000
      - 
        key: Last system time (s)
        value: 0.000000000
      - 
        key: Time offset (s)
        value: 0.000000000
---

`
 In Level 1, it says 'not connected'.  
How can I fix this connection problem??  
I tried to set baudrate as 57600, 921600 and 1152000 but it did not work.

+ I checked out that the connection between pixhawk and my desktop is fine. Only the connection between pixhawk and TX1 is the problem. 
Any advice will be very thankful to me.

mhkabir, 2016-09-18:
It's your hardware issue. We can't really help with that.

mhkabir, 2016-09-18:
Add your user to the dialout group and remove modemmanager. That should fix it.

yunshin, 2016-09-18:
Thanks for reply mhkabir!
Do you mean TX1 when you say hardware??  
I checked out that the normal communication using ROS without mavlink works fine in TX1. Also I set fcu_url as both local host( 127.0.0.1 with gazebo) and serial communication to check if the problem is related with serial communication system of TX1. However both of them do not work. I think it is related with mavlink or mavros. Maybe I wrongly installed some packages.. How do you think about that?    
 + I'll add my user to the dialout group as you advice and post the result.

mhkabir, 2016-09-18:
It's got nothing to do with mavros or mavlink. Your serial connection isn't setup properly either way. 

Please note that we will also not support any issues you may have while using a USB connection to the flight controller. It is not recommended as a flight interface.

kyw2539, 2016-09-18:
I am having same problem with you yunshin. Do you solve that problem?

yunshin, 2016-09-19:
@mhkabir 
I removed modemmanager by cliking 'remove' buttion in Ubuntu Software Center and add my user to the dialout group by run : $sudo usermod -a -G dialout $USER.
However it does not seem to work.. I think I have to change connection protocol. 
Do you mean to use uart( TELEM2 port on pixhawk) to USB( on TX1) connection?? 
Thanks!

mhkabir, 2016-09-19:
http://dev.px4.io/pixhawk-companion-computer.html

closed 2016-09-19 by mhkabir