mavlink/mavros #605: RCIN sampling rate

Muttle, 2016-09-24:
Hi all,

I am trying record certain published topics using rosbag, whilst the drone is flying. 

This works great except my my rcout topic sampling rate drops down to 4Hz once the drone begins to fly. Really, I'd like to sample this signal at 10Hz if possible.

Is this a problem inside my FCU (pixhawk px4) or can this be set elsewhere?

Thanks in advance.

mhkabir, 2016-09-25:
What's your flightstack? 
How are you connecting to the vehicle?

Muttle, 2016-09-25:
Hi my flight stack is apm2 as it is a conventional  helicopter. Communication is via telem2 at custom baud rate @ 500000bps.

mhkabir, 2016-09-25:
It's likely a comms/flightstack issue. Unlikely to be comms, but could be slowdowns in the ROS transport if the CPU load is too high. Are you starting anything on the companion during flight? 

I suggest speaking with the APM developers otherwise.

Muttle, 2016-09-25:
I managed to update the signal to about 10Hz via mission planner in the telemetry rates. I am running another package which takes vicon data but nothing much really. Just using it for logging purposes only.

closed 2017-01-23 by vooon