mavlink/mavros #606: (0.18.3) ATT_POS_MOCAP messages not being sent, no data published to /mocap/tf

JGSOpenSrc, 2016-09-27:
I have been unable to use the mocap position estimation plugin. I'm using mavros in conjunction with the [vicon_bridge](https://github.com/ethz-asl/vicon_bridge) node from ethz to send mocap data to a quadrotor using the PX4 flight stack.

I tried editing px4_config.yaml to set the use_tf parameter to true, but this had no effect. I have verified that my bridge node is publishing the data to the /tf topic.

Interestingly, I am able to send VISION_POSE_EST messages using the vision_pose plugin. Am I missing some part of the launch configuration? How do I properly setup the mocap plugin?

vooon, 2016-09-27:
Please note that mocap plugin is experimental and i do not have reports on vicon integration.

As i understood that bridge send data in TF messages, but plugin do not use TF lib.
It subscribe to some topic with TF message type. So you must remap /tf of vicon to plugin.
And check that vicon send only one transformation. If not, you must edit plugin to use TF lib to get right transformation to send.

JGSOpenSrc, 2016-09-27:
Thanks Voon!

Should I just stick to using the vision_pose plugin instead? I'm having a lot of difficulty getting basic position control using vision_pose.

I've been relying on unmodified mavros and vicon_bridge nodes, straight from the git repositories.

vooon, 2016-09-28:
Maybe. Anyway you must ensure that FCU support that position source message, find how to get position data from mocap system, connect it to mavros plugin. Because there are dozens of variations you may has to write your own plugin (but that should be simple).

genemerewether, 2016-10-03:
Which PX4 estimator are you using, and which hardware? And what is the output from the pxh> shell of:
listener att_pos_mocap

That will tell you whether mocap messages are getting into the PX4 flight stack.

TSC21, 2017-04-04:
@JGSOpenSrc is this solved to you?

TSC21, 2017-04-14:
Closing this as there's no feedback. Please reopen if needed.

closed 2017-04-14 by TSC21