mavlink/mavros #608: Question about topic mavros/imu/atm_pressure

JM-SNU, 2016-09-29:
Hello, all.

I have some problems with obtaining pressure information from pixhawk. 

I use qgroundcontrol software to set "pixhawk", and receive sensor data (e.g. gyro, acc, baro, ... ) using "mavros" package in ROS.

When I use 2.7.1 version, I can receive air pressure value from pixhawk (ROS topic is mavros/imu/atm_pressure).

However, after updating firmware and setting pixhawk using 3.0.1 version, the air pressure value is always zero.

That means, I can subscribe topic about air pressure but cannot receive correct pressure data.

I'm not sure, but I think it is problem with version of qgroundcontrol. (or parameter setting in updated version) or mavros package (other topics like mavros/imu/data, mavros/imu/mag  are obtained correctly)

And, I tried to install updated mavros package, but I cannot install correctly, so now I use old version of mavros package (maybe I installed it on April, 2016)

If you think it is problem with mavros package (like version,..) , let me know how I can handle this problem.
( And if it is problem with qgroundcontrol, maybe I should ask them it)

Thank you.

vooon, 2016-09-29:
a. What firmware (flight stack) are used? PX4 / APM? APM copter (which)?
b. `rostipic echo -n1 /disgnostics` in \`\`\` code block
c. How you launch node? Log output may be useful too
d. Did you try to change message streams rate? `mavsys rate --help`
e. What the problem with update? Did you try to [follow instructions][1]? What ROS version used? OS version?

[1]: https://github.com/mavlink/mavros/tree/master/mavros#installation

JM-SNU, 2016-09-29:
a. I use PX4 Flight stack

b. I tried, and results is like this.
header: 
  seq: 144
  stamp: 
    secs: 1475141287
    nsecs: 629067252
  frame_id: ''
status: 
  - 
    level: 0
    name: mavros: FCU connection
    message: connected
    hardware_id: /dev/ttyUSB0:921600
    values: 
      - 
        key: Received packets:
        value: 56946
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 39
      - 
        key: Tx sequence number:
        value: 1
      - 
        key: Rx total bytes:
        value: 2331720
      - 
        key: Tx total bytes:
        value: 43008
      - 
        key: Rx speed:
        value: 19142.000000
      - 
        key: Tx speed:
        value: 360.000000
  - 
    level: 0
    name: mavros: GCS bridge
    message: connected
    hardware_id: /dev/ttyUSB0:921600
    values: 
      - 
        key: Received packets:
        value: 0
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 0
      - 
        key: Tx sequence number:
        value: 0
[... 144 lines cut ...]
[ INFO] [1475141105.617150974]: Plugin vision_pose_estimate loaded and initialized
[ INFO] [1475141105.621955895]: Plugin vision_speed_estimate loaded and initialized
[ INFO] [1475141105.627430012]: Plugin waypoint loaded and initialized
[ INFO] [1475141105.627572355]: Built-in SIMD instructions: SSE, SSE2
[ INFO] [1475141105.627591971]: Built-in MAVLink dialect: ardupilotmega
[ INFO] [1475141105.627603467]: MAVROS started. MY ID [1, 240], TARGET ID [1, 1]
[ WARN] [1475141105.711080657]: GP: No GPS fix
[ INFO] [1475141106.306266415]: CON: Got HEARTBEAT, connected.
[ INFO] [1475141106.313138258]: IMU: High resolution IMU detected!
[ INFO] [1475141106.321958757]: RC_CHANNELS message detected!
[ INFO] [1475141107.316368499]: VER: 1.1: Capabilities 0x00000000000004eb
[ INFO] [1475141107.316683761]: VER: 1.1: Flight software:     01040100 (0c2aa165feee1cb4)
[ INFO] [1475141107.316857261]: VER: 1.1: Middleware software: 01040100 (0c2aa165feee1cb4)
[ INFO] [1475141107.317012909]: VER: 1.1: OS software:         ffffffff (0000000000000000)
[ INFO] [1475141107.317126564]: VER: 1.1: Board hardware:      00000011
[ INFO] [1475141107.317236919]: VER: 1.1: VID/PID: 26ac:0011
[ INFO] [1475141107.317403194]: VER: 1.1: UID: 3333471335313431
[ INFO] [1475141121.312279485]: WP: mission received

d.  I enter like "rosrun mavros mavsys rate --all 10", but nothing's changed...

e. I don't know..  I usually download mavros package using git clone and catkin_make, but the instruction is not same. So I don't know how I can install mavros package using this method. ( I tried source installation method, but doesn't work -- catkin build or make doesn't work)
And ROS version is indigo / OS version is Ubuntu 14.04 

Thanks.

vooon, 2016-09-29:
I do not see any problem in diag & log. To run better to use `roslaunch mavros px4.launch fcu_url:=<url>` since it will load some defaults.

d. Perhaps only work on APM.

Atmospheric pressure is a part of HIGHRES_IMU message, and i see log message about it's detection.
Check firmware side, and ensure that bit mask still valid.

e. Indigo too outdated, but current mavros still may work on it. Use `--rosdistro kinetic` on both calls.
Version in Indigo repo too outdated to compile. So you should use Kinetic.

What error give wstool?

LorenzMeier, 2016-09-29:
@JM-SNU Can you connect to QGroundControl and look at the highres IMU message?

@vooon PX4 implements the altitude message:
http://mavlink.org/messages/common#ALTITUDE

vooon, 2016-09-29:
@LorenzMeier and it is supported by [altitude plugin][1] :)

[1]: https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/altitude.cpp#L56

JM-SNU, 2016-09-30:
@vooon I tried using "roslaunch mavros px4.launch fcu_url:=/dev/ttyUSB0:921600" but results are same.. air pressure value is still zero.

d. OK. And How can I check firmware side? I already tried to calibrate sensor, set parameter or upload firmware, etc.. but it doesn't work.. I'm curious that other IMU topics (mavros/imu/data, mavros/imu/mag,...) are obtained correctly, but the only mavros/imu/atm_pressure data is not available..

e.  As I remember, error is about "2016.9.9 version check". 

@LorenzMeier I tried analyzing data using qgroundcontrol. Then, other IMU values are obtained correctly. (e.g. x/y/z acc/gyro/mag) However, abs_pressure is still zero. I think it is problem with firmware from the qgroundcontrol...   
But the strange thing is that pressure_alt value has non-zero value but abs_pressure is always zero. Do you know why? Thanks.

vooon, 2016-09-30:
e. Try to remove `mavros` and `mavlink` folders, then edit `ws/src/.rosinstall` file to remove that entries.

vooon, 2016-09-30:
Anyway for what do you want pressure data?

JM-SNU, 2016-09-30:
@vooon  Thanks, I will try to install mavros following your advice.
And I want pressure data to obtain z-position measurement using raw-sensor data (not using pixhawk estimator) Thanks.

hridaybavle, 2016-09-30:
Hi, I was having the same problem, that after updating the firmaware of pixhawk to 1.4.4, it stopped publishing atm pressure, for now I went back to 1.3 version to get the pressure data again. 

Let me know if you find the solution.

closed 2017-03-03 by vooon