mavlink/mavros #610: Problem with using Optitrack to control quad based on mavros

luxiangnk, 2016-10-06:
Hi all:

Our lab have a set of optitack and I want to do some quadrotor control with this motion capture system. What I have done was:

1- Send the position of quad to mavros with coordinates in ENU
2- Feed positions to Pixhawk from mocap in NED
3- I choose position inav in QGC
4- I send the position and orientation to the mavros topic "/mavros/mocap/pose"
4- I got the message of "FCU :[inav] MOCAP data valid", and I also can get the mocap postion from the SD card of quad.

And then I have done(use offboard demo from "https://px4.osdrone.net/10_Robotics-using-ROS/ros-mavros-offboard.html"):
1- I armed the quad
2- I activated the offboard mode
3- I send position setpoints through sending mavros conmand(/mavros/setpoint_position/local)

Now, the quad can take off. However, the quad cannot hold and go in a weird direction and crashed.

Were there any problems with versions of mavros or Firmware?
Were there any problems with the setup of QGC, which params should I set？

I follow the https://px4.osdrone.net/10_Robotics-using-ROS/external-position.html to set the QGC.

Could anyone can help and tell me which part
I have done was wrong? Thanks so much for the help!

Jaeyoung-Lim, 2016-10-28:
Have you checked the coordinates of the mocap data?

I feel like if you are sending the pose, you should send it in ENU

cookz2, 2016-12-20:
I've also had the same sort of problem. Any updates Luxiangnk?

weiweikong, 2016-12-20:
@cookz2 Which PX4 Firmware version did you choose?

cookz2, 2016-12-21:
@weiweikong It was the *current * version about a month ago when I pulled it from git. I'm not sure about the exact version but i can find out today.

weiweikong, 2016-12-22:
@cookz2 Thank you for your reply. Have you encounter the similar situation as shown in issuehttps://github.com/PX4/Firmware/issues/6106?

cookz2, 2016-12-22:
@weiweikong  We have been constraining the Quad physically and trying to see response since it seemed to be flying in a weird direction right from the start when we enter off board mode and it  never entered a stable flight period.  Which coordinate frames are you sending its reference position and actual position from motion capture?

weiweikong, 2016-12-23:
@cookz2 We are using VICON system with ETH `vicon_bridge` package and the coordinate of VICON is `ENU`, whose East is not aligned the real east, but it is the standard right-hand coordinates, whose x is the right, y is the forward and z-axis points to the air. So the VICON only send the ENU data to PX4 though `mavros/mocap/pose` and the PX4 was configured using `LPE estimator`, and we disabled the GPS and Vision related parameters.

In my experience, but not confirmed from the PX4 team that, according to the position control module for multi-copter, **there is a minimum take-off speed**, `tko_speed`, which seems to make sure that the multi-copter is in the air before we switch to OFFBOARD. So this safety checking logic results in an aggressive maneuver as shown in the log, http://logs.uaventure.com/view/PTMjRcANNg6y5fex4SU8Aj. The detail of the logged flight was
- Set the quadrotor on the ground and sent the `VICON ENU` data to `PX4`.
- Manually ARM the quadrotor through RC and the rotors were spinning in idle speed.
- Manually switched the OFFBOARD channel through RC and the quadrotor **jumped** into the air which is caused by `tko_speed`.
- When reaching the desired `tko_speed`, the quadrotor could be controlled normally in OFFBOARD mode as can be seen in the rest section of the log diagrams.

Currently, it is still a problem that the yaw angle was not stable, which might be caused by the fake ENU coordinate. Although we set the `ATT_EXT_HDG_M = 2`, the heading problem still exist.

cookz2, 2016-12-23:
@weiweikong Thanks so much for your response. I will try that today and see if i experience the same sort of jump at takeoff before proceeding to the actual position control.

weiweikong, 2016-12-24:
@cookz2 Great! Wait for your result. :)

cookz2, 2016-12-26:
@weiweikong So we got it flying on Friday and I think you are completely right about the minimum velocity needed to enter real offboard Flight.  Initially I thought there was a problem with what we were doing so we ( stupidly ) had somebody hold the bottom of the quad and at least try to see where it was trying to go. This stopped it from reaching the needed velocity and consequently we never entered offboard mode, so we assumed we had some transforms wrong. After just letting the quad fly by itself  it popped up and flew fine but I do believe you are right for that minimum velocity needed. 

Do you happen to have a heavy payload or an underpowered quad? This might be what's making it slow to enter the offboard mode in the issue you linked earlier. Also, thanks for all your help!

weiweikong, 2016-12-27:
@cookz2 It's nice to hear the results from you.  I checked the `mc_pos_control.cpp` which has a section related the `tko_speed`. I opened an issue (https://github.com/PX4/Firmware/issues/6106) related to this but did not get confirmed reply yet. Would it be possible to update you log online and let the PX4 guys know this result exists?

The first attempts of OFFBOARD testing are quite a challenge, and we also did what you have tried to detect where the vehicle tries to go. Confused by the maneuver, we checked log and the source code almost confirmed the minimum velocity exists. We tuned the `tko_speed` parameter to `1` and found the quad took off more slowly. Currently, we are considering to remove this section or set the `tko_speed` much smaller to control it taking off slowly in the OFFBOARD mode.

Previously, the `tko_speed` is relatively high to our platform, so it boosted up quickly and struck the ceiling heavily. So as you mentioned, if the payload is heavy, it might persist fly into the sky.

weiweikong, 2016-12-28:
@cookz2 which PX4 firmware version did you choose for your last testing? I would like to compare the firmware version and fix the `tko_speed` constriction.

TSC21, 2017-04-04:
@luxiangnk and @cookz2 is this solved to you both?

TSC21, 2017-04-14:
Closing this as there's no feedback. Please reopen if needed.

closed 2017-04-14 by TSC21

EdXian, 2017-08-11:
@luxiangnk hi , had you solved the problem ? if yes, can you shared with me in more detail?