mavlink/mavros #616: Problems with vision_pose/pose

NCharabaruk, 2016-10-21:
Hello,

I am using a Pixhawk with PX4 firmware for indoor navigation with a hexacopter being controlled by ROS. I am using laser scan matching to provide xy position data and yaw, a sonar sensor for z, and an IMU for roll and pitch. This data is put into a single message and published on /mavros/vision_pose/pose. When looking at /mavros/local_position/pose however, the output is different. The x and y are similiar, but the z is off by a large amount. Does anyone know how to fix this?

Thanks

AlexisTM, 2016-12-14:
Dear,

Could you double check the information you send via /mavros/vision_pose/pose ? (rqt_plot can help here)

```
rqt_plot /mavros/vision_pose/pose/pose/position/x
```

mhkabir, 2017-02-24:
Closing as stale.

closed 2017-02-24 by mhkabir