mavlink/mavros #617: The mavros/setpoint_attitude/attitude topic not connected to /mavros node

WrightFlyerZhou, 2016-10-24:
Hi, I have a few nodes that I want to use a joystick to control a quadrotor, so I tried to use both /mavros/setpoint_attitude/att_throttle and mavros/setpoint_attitude/attitude topics. However, I see only mavros/setpoint_attitude/att_throttle topic is connect to /mavros node, see the [figure](https://lh3.googleusercontent.com/-AnoI5fhOTOc/WA6ZGpfoH5I/AAAAAAAAGt8/u7Eu1qyPvGMin-sFDUyKdqLRhxVTGsJiwCLcB/s1600/rqt.jpg) here. 
Anyone has any idea?

Thanks

vooon, 2016-10-25:
Use rosnode!

closed 2016-10-25 by vooon

WrightFlyerZhou, 2016-10-28:
What do you mean? Why you close it? rqt_graph is not a correct tool????

DINGJIANG ZHOU
Department of Mechanical Engineering
Boston University

vooon, 2016-10-28:
Because rqt_graph is incorrect tool to validate node topics. `rosnode` can show all topics: publisher, subscriber *and* connections.

And i believe if you stop rostopic subscriber you will see connection to mavros.

WrightFlyerZhou, 2016-10-28:
Thanks for your answer. I appreciate.

Dingjiang

DINGJIANG ZHOU
Department of Mechanical Engineering
Boston University