mavlink/mavros #620: custom_mode: OFFBOARD invalid?

zhahaoyu, 2016-10-27:
After I connected my ros node to mavros, I was trying to send custom_mode OFFBOARD, but the mavros console outputs `[ WARN] [1477604919.660821416]: CMD: Unexpected command 176, result 0
`

The code I use is 

   ```
    ros::ServiceClient set_mode_client = nh.serviceClient<mavros_msgs::SetMode("mavros/set_mode");
    mavros_msgs::SetMode offb_set_mode;
    offb_set_mode.request.custom_mode = "OFFBOARD";
    set_mode_client.call(offb_set_mode)
```

vooon, 2016-10-28:
And what? State.custom_mode won't change to offboard?

vooon, 2016-10-28:
And i guess you use PX4? Versions? ROS?

vooon, 2016-10-28:
Related / dup of #512

zhahaoyu, 2016-10-28:
@vooon Yes I am using PX4, latest source version and ROS kinetic. What I found out is

1. I have to arm the vehicle, then switch it to offboard, which is different from the order provide by the tutorial [http://dev.px4.io/ros-mavros-offboard.html](url)
2. I am able to change it to offboard mode, but mavros still reports it as unexpected command.

vooon, 2016-10-28:
You may ignore command 176 and 400 because it's parasitic effect of PX4 internal message processing.
PX4 resend SET_MODE message to uORB command topic (also it serve COMMAND_LONG and _INT)
so command processor later send confirmation which is unexpected for command plugin.

zhahaoyu, 2016-10-28:
Thanks @vooon

closed 2016-10-28 by zhahaoyu

bhaskar-glitch, 2021-11-24:
> You may ignore command 176 and 400 because it's parasitic effect of PX4 internal message processing. PX4 resend SET_MODE message to uORB command topic (also it serve COMMAND_LONG and _INT) so command processor later send confirmation which is unexpected for command plugin.

HI @vooon I need a help..I'm unable to switch to offbaord mode and the error is **CMD: Unexpected command 176, result 1**
please help