mavlink/mavros #623: Controlling px4 with joystick using setpoint_velocity/cmd_vel

L4ncelot, 2016-11-01:
Hi,

I've been playing with **px4** firmware which I connected with **ROS** and **Gazebo**. What I want to achieve right now is controlling my copter with **joystick** for which I use **teleop.launch** from **mavros_extras** package. I'm able to control my copter by sending **geometry_msgs/TwistStamped** to **mavros/setpoint_velocity/cmd_vel** topic. 

The problem is that it seems like this message is related not to current body frame of my copter rather than to the world coordinates. Which means that when I want to fly forward in e.g. north direction, I have to send positive values of pitch. But when I want to fly forward in south direction, I have to send negative values of pitch to do so (I'm not sure where the north is right now, it's just an example of what's going on).

So my question is if there's a way to send velocity messages in respect of **body_frame** rather than **world_frame** and if so then how do I use it.

Thank you for any advice.

Jaeyoung-Lim, 2016-11-01:
The standard ROS frame is based on ENU frame and the cmd_vel topics are based on the inertial frame

You can use the setpoint attitude topics to directly control the attitude

L4ncelot, 2016-11-01:
Oh, thanks, that's what I wanted.

closed 2016-11-01 by L4ncelot

vinhk, 2017-10-19:
@Jaeyoung-Lim would you able to share your complete code for attitude control?

L4ncelot, 2017-10-20:
@vinhk if you want you can write it yourself, it's not that hard. Make some PID control loop with e.g. desired position. Read values about local position of your copter and give errors to your PID controller. Don't forget to use **OFFBOARD** mode. [Here](https://dev.px4.io/en/ros/mavros_offboard.html) is the link on **OFFBOARD** tutorial.

L4ncelot, 2017-10-20:
Also don't forget when changing to **OFFBOARD** mode you need to actually publish some messages to **setpoint_attitude** topic. Otherwise the switch will be denied.

vinhk, 2017-10-20:
Do you have a code for publishing setpoint_attitude topic?

L4ncelot, 2017-10-23:
Hi,

here's my very old code for simple attitude control. mavros_tools is just
some library I've written which takes care of publishing and subscribing to
some mavros topics. I don't know if the code works or not, but it should
give you a hint how to write your own controller. Hope it helps.

Have a nice day.