mavlink/mavros #624: How to build FCU <--> mavros <--> QGC chain

zhahaoyu, 2016-11-01:
Hi, if I plug the telemetry into my computer, and open a mavros node to communicate with the FCU via the telemetry. What UDP address do I put when I launch mavros?

weiweikong, 2016-11-02:
- The UDP address is based on your PC IP address and related port. You could set the address in your `launch file` or on the command line when your run it. 
- The basic structure is 
```
roslaunch mavros px4.launch fcu_url:="udp://:14540@192.168.1.36:14557"
````
however the port and address are depends on your system.
- Please refer http://dev.px4.io/simulation-ros-interface.html.

zhahaoyu, 2016-11-02:
thanks. I understand this, but when I plug in the telemetry onto my computer, what IP address do I use. Where can I find the IP address of the FCU? I am using rc telemetry for transmission

weiweikong, 2016-11-02:
@zhahaoyu You could check your PC IP address using `ifconfig`. 
- If you using telemetry method, `mavros` have already received `mavlink` message from FCU(PX4) through telemetry port and transferred them to ROS message. You just let the `mavros` do another thing, to publish the same information into the UDP port, where QGC could receive the information. 
  - Connection between FCU and your PC

    \<FCU\> ---telemtry port --\<MAVROS\>

  - In your PC:

    \<MAVROS\> --- UDP ----\<QGC\>
    \<MAVROS\> ----ros message---\<Other Nodets\>

- So, there is no need know the IP address of FCU, and there is no IP address for FCU actually.

zhahaoyu, 2016-11-02:
Thanks @weiweikong. Now I understand the structure. I tried to use the `roslaunch mavros px4.launch`, and I am able to start a Mavros node communicating with FCU via telem, but how do I open a UDP port for QGC, and instruct QGC to listen to it?

When I start Mavros, everything was fine. But if I open QGC, then it automatically listens to the same telem port, and mavros crashes. So how can I create the connection between QGC and mavros

weiweikong, 2016-11-02:
@zhahaoyu 
- For UDP configuration
```
<include file="$(find mavros)/launch/node.launch">
		<arg name="blacklist_yaml" value="$(find mavros)/launch/px4_blacklist.yaml" />
		<arg name="config_yaml" value="$(find mavros)/launch/px4_config.yaml" />

		<arg name="fcu_url" value="$(arg fcu_url)" />
		<arg name="gcs_url" value="$(arg gcs_url)" />
		<arg name="tgt_system" value="$(arg tgt_system)" />
		<arg name="tgt_component" value="$(arg tgt_component)" />
	</include>
````
the `gcs_url` section is for the UDP configuration. The old px4 website has more detail https://pixhawk.org/peripherals/onboard_computers/intel_edison.
As the example shows
````
<arg name="gcs_url" default="udp://:14556@my-intel-edisons-ip:14550" />
```
means QGC will then be connected via UDP to port 14556.
- For safety, you could check the UDP ports to make sure it was available or occupied.

zhahaoyu, 2016-11-02:
I followed your instruction, here is the launch file I used

```
<launch>
	<!-- vim: set ft=xml noet : -->
	<!-- example launch script for PX4 based FCU's -->

	<arg name="fcu_url" default="/dev/ttyUSB0:57600" />
	<arg name="gcs_url" default="udp://:14560@127.0.0.1:14550" />
	<arg name="tgt_system" default="1" />
	<arg name="tgt_component" default="1" />
	<arg name="log_output" default="screen" />

	<include file="$(find mavros)/launch/node.launch">
		<arg name="pluginlists_yaml" value="$(find mavros)/launch/px4_pluginlists.yaml" />
		<arg name="config_yaml" value="$(find mavros)/launch/px4_config.yaml" />

		<arg name="fcu_url" value="$(arg fcu_url)" />
		<arg name="gcs_url" value="$(arg gcs_url)" />
		<arg name="tgt_system" value="$(arg tgt_system)" />
		<arg name="tgt_component" value="$(arg tgt_component)" />
		<arg name="log_output" value="$(arg log_output)" />
	</include>
</launch>

```

and mavros still crashes with following error message

```
ERROR] [1478057788.915106802]: serial0: receive: End of file
terminate called after throwing an instance of 'std::system_error'
  what():  Resource deadlock avoided
================================================================================REQUIRED process [mavros-2] has died!
process has died [pid 12427, exit code -6, cmd /opt/ros/kinetic/lib/mavros/mavros_node __name:=mavros __log:=/home/haoyu/.ros/log/849342f6-a0ad-11e6-8d09-000f6005d13e/mavros-2.log].
log file: /home/haoyu/.ros/log/849342f6-a0ad-11e6-8d09-000f6005d13e/mavros-2*.log
Initiating shutdown!
================================================================================
[mavros-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

weiweikong, 2016-11-02:
@zhahaoyu I am not quite sure about this situation. In my experience, QGC should work independently with ROS, so it should be start-up normally, and we could set QGC UDP configuration to listen to the information from `mavros`. 
```
serial0: receive: End of file
```
I am now sure whether above information means QGC tried to connect FCU with serial port. If it was true, QGC or `mavros` might crashed.

zhahaoyu, 2016-11-02:
I believe QGC is trying to use the serial port. When I use QGC with ROS in SITL mode, everything works fine. I just don't know how to do this with real pixhawk fcu.

weiweikong, 2016-11-02:
@zhahaoyu Maybe try to start QGC first then configure it using UDP connection instead of serial port before you launching the mavros wih PX4.

zhahaoyu, 2016-11-02:
@weiweikong  Thanks so much. I solved the problem. For others' reference, here is what I did:

1.  Open QGroundControl Station, and click the "Q" icon on the top left, entering "Preferences" panel
2.  Click "General" on the left, then find "Autoconnect to the following devices:", opt out "SiK Radio", and opt in "UDP"
3.  Close QGroundControl to ensure the it is disconnected from the FCU
4.  Edit the px4.launch file, which is found in the Mavros package folder, add "udp://:14560@127.0.0.1:14550", 14560 and 14550 are any unused ports
5.  Run `roslaunch mavros px4.launch`
6.  Open QGroundControl

Now a chain FCU <-telemetry-> MAVROS <-UDP-> QGC is established. QGC will not automatically connect with the FCU vai Telemetry anymore. If we still want that, we could configure a manual connection.

closed 2016-11-02 by zhahaoyu

vooon, 2016-11-02:
FYI: Try `udp-b://@` as GCS URL.

m1577, 2017-05-05:
Hi, I have a similar problem, I got an FPV Radio Telemetry Ground Module and I want it to communicate with the telemetry that owns the Iris drone.

When I connect the telemetry (Ground) to my ubuntu pc and turn on the drone does not start the Qground station.

Why is this problem?

Do I have to install any drivers?