mavlink/mavros #626: libmavconn not compiling on OSX

pvechersky, 2016-11-15:
libmavconn is failing to compile on OSX due to an error in thread_utils.h (https://github.com/mavlink/mavros/blob/master/libmavconn/include/mavconn/thread_utils.h). This appears to be due to an issue with the different APIs for pthread_setname_np (http://stackoverflow.com/questions/2369738/can-i-set-the-name-of-a-thread-in-pthreads-linux/7989973#7989973).

Would it be possible to extend libmavconn to support OSX?

Thank you!

vooon, 2016-11-15:
It is possible, and relatively easy, but i don't has any OSX machine where i can check changes.

TSC21, 2017-04-04:
@pvechersky are you able to compile it already on MAC OS X?

pvechersky, 2017-04-05:
@TSC21 unfortunately I don't have access to an OSX machine so I am not able to test this :(

TSC21, 2017-04-05:
Is there anyone with a OSX machine that can test this?

pvechersky, 2017-04-05:
I believe that @ffurrer has access to an OSX machine but could be wrong

TSC21, 2017-04-06:
@ffurrer would you be kind to test this on your OSX? Thanks

TSC21, 2017-06-09:
@pvechersky can this be closed now?

closed 2017-06-10 by pvechersky