mavlink/mavros #628: Subscribing to mavros messages

donald2016, 2016-11-23:
Hi,

I am trying to communicate Pixhawk with Raspberry Pi. In Pixhawk, the firmware is Arducopter 3.3.2. In Raspberry Pi, the OS is Ubuntu mate 16.04 and ROS Kinetic is installed.

On Raspberry Pi, I have run the following commands in four terminals separately
`
roscore
roslaunch mavros px4.launch
rosservice call /mavros/set_stream_rate 0 10 1
rostopic echo /mavros/imu/data
`
The last terminal gave me the data I need as follows
`
header: 
  seq: 58711
  stamp: 
    secs: 1479869629
    nsecs: 975670641
  frame_id: fcu
orientation: 
  x: 0.0461782780217
  y: 0.00985117873358
  z: -0.0963877134008
  w: -0.994223279561
orientation_covariance: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
angular_velocity: 
  x: 0.000993488705717
  y: -0.100484438241
  z: 0.00226617511362
angular_velocity_covariance: [1.2184696791468346e-07, 0.0, 0.0, 0.0, 1.2184696791468346e-07, 0.0, 0.0, 0.0, 1.2184696791468346e-07]
linear_acceleration: 
  x: 0.06864655
  y: 0.2745862
  z: 9.7281968
linear_acceleration_covariance: [8.999999999999999e-08, 0.0, 0.0, 0.0, 8.999999999999999e-08, 0.0, 0.0, 0.0, 8.999999999999999e-08]
`

My next step is to create a node which needs the above imu data for further calculation. However, I cannot subscribe to read the imu data. 
`
#include <ros/ros.h>
#include <geometry_msgs/PoseStamped.h>
#include <mavros_msgs/CommandBool.h>
#include <mavros_msgs/SetMode.h>
#include <mavros_msgs/State.h>
#include <geometry_msgs/PoseStamped.h>
#include <geometry_msgs/Pose.h>
#include <sensor_msgs/Imu.h>
#include <mavros_msgs/StreamRate.h>

mavros_msgs::State current_state;
void state_cb(const mavros_msgs::State::ConstPtr& msg){
    current_state = *msg;
    printf("connected %i \n",current_state.connected);
    printf("armed %i \n",current_state.armed);
}

geometry_msgs::PoseStamped current_orientation;
void imu_cb(const geometry_msgs::PoseStamped::ConstPtr &imureq){
    current_orientation = *imureq;
    printf("sensor %f\n",current_orientation.pose.orientation.x);
}

int main(int argc, char **argv)
{
    ros::init(argc, argv, "offb_node");
    ros::NodeHandle nh;
    ros::ServiceClient client = nh.serviceClient<mavros_msgs::StreamRate>("/mavros/set_stream_rate");
    mavros_msgs::StreamRate srv;
    srv.request.stream_id = 0;
    srv.request.message_rate = 10;
    srv.request.on_off = 1;
    if (client.call(srv)){
            ROS_INFO("Service Called");
    }else{
            ROS_INFO("Failed to call service");
    }

    ros::Subscriber state_sub = nh.subscribe<mavros_msgs::State>
            ("mavros/state", 10, state_cb);
    ros::Publisher local_pos_pub = nh.advertise<geometry_msgs::PoseStamped>
            ("mavros/setpoint_position/local", 10);
    ros::ServiceClient arming_client = nh.serviceClient<mavros_msgs::CommandBool>
            ("mavros/cmd/arming");
    ros::ServiceClient set_mode_client = nh.serviceClient<mavros_msgs::SetMode>
            ("mavros/set_mode");
    ros::Subscriber imu_data_sub = nh.subscribe<geometry_msgs::PoseStamped>
            ("mavros/imui/data",10,imu_cb);
    ros::Rate rate(20.0);
    ros::spin();
    return 0;
}
`

As shown in the code, this node subscribes and provides the correct data for mavros/state in 'armed' and 'connected'. However, it does not subscribes to read the data from mavros/imu/data. The statement to display the orientation never appear.
Would you mind please give me a hand to read the mavros/imu/data?

Regards,
Donald

closed 2016-11-23 by donald2016

vooon, 2016-11-23:
IMU topic is not `geometry_msgs::PoseStamped`, it is [`sensor_msgs::Imu`][1].

[1]: http://docs.ros.org/api/sensor_msgs/html/msg/Imu.html

donald2016, 2016-11-23:
Thanks Voon I figured out this mistake after I post this thread. Thanks