mavlink/mavros #629: Steps for OverrideRCIn

donald2016, 2016-11-28:
I am trying to set up the OverrideRCIn from Raspberry Pi 2 to Pixhawk by Tele 2
Current my Pixhawk is APM 3.3.2 Arducopter
I can listen to the rc/in and rc/out by
using apm2.launch, setup the stream rate, then rostopic echo /mavros/rc/in and /mavros/rc/out

I start up a fourth terminal to setup SYSID_MYGCS by
```
$ rosrun mavros mavparam set SYSID_MYGCS 1
1
$rosrun mavros mavparam get SYSID_MYGCS
1
```
Then I use the following command to execute the override
```
rostopic pub -1 /mavros/rc/override mavros_msgs/OverrideRCIn '[1500,1500,1900,1500,1325,1560,1500,1500]'
```
But the rc/in and rc/out terminals did not change after the override command. Moreover, I can still gain control manually.

I have read some thread, using OverrideRCIn need to setup the correct target. Did I miss anything on setting up the system for override?
Do I have to set up the flight mode to get ready for override? I have tried to input the command during "Stabilize", "Auto" and "Guided" mode. It gave me no response.

vooon, 2016-11-28:
Override works in any mode (it's low level). MYGCS should be same as mavros sysid. Try to use checkid script.
Also 3.3.2 released in 2015, try to update.

donald2016, 2016-11-29:
Thanks Voon.
I have updated the firmware to 3.4.2.
I got the following result by checkid
```
$ rosrun mavros checkid
OK. I got messages from 1:1.

---
Received 2923 messages, from 2 addresses
sys:comp   list of messages
 252:1     0, 66
  1:1     0, 1, 2, 193, 152, 150, 24, 27, 29, 30, 33, 35, 36, 165, 42, 178, 182, 62, 65, 74, 163, 241, 116, 125
```
I tried both 252 and 1 of SYSID_MYGCS. There were no response.

I found another command to check the /mavros/system_id. However, the parameter was not set yet

```
$ rosparam get /mavros/system_id
ERROR: Parameter [/mavros/system_id] is not set
```

how do I set up the sysid for mavros?

vooon, 2016-11-29:
Since that type of control not suitable for automatic control, i did not check that for long time.
Check APM receiver code (maybe there silently drops message (msg crc)?), try to stream continuous (try sin). Try to set only one port, all other leave = 0.

vooon, 2016-11-29:
Also try DroneKit, does override work with it?

donald2016, 2016-11-30:
If OverrideRCIn is not suitable for automatic control, what is a better method to set the UAV attitude with APM? Should I use setpoint_raw to control the UAV attitude instead of OverrideRCIn ? I have read your other reply mentioned that APM only support setpoint_raw.

vooon, 2016-11-30:
Yes, setpoint_raw is what you need. But ensure what set of data APM expect.

donald2016, 2016-12-01:
Please correct me if I am not correct.
As I know, to control setpoint_raw is only avaliable in GUIDED mode for APM. However, I am running experiments indoor that I cannot receive GPS signal. Thus, I cannot arm the system even I disable the arming check. Are there any method to get around it?

vooon, 2016-12-01:
There should be a setting that allow to arm without GPS. Ask support at APM forum.
Should be possible to stream position from external source.
Yes, setpoint works in GUIDED.

donald2016, 2016-12-05:
I have solved the problem about OverrideRCIn which was a simple problem.
I read the raw mavlink message. I found that the Pixhawk system id and component id are different from the setting from mavros. Thus, I changed the target ids and everything works fine. I also disabled the GPS requirement from APM and make the setpoint_raw available.

vooon, 2016-12-05:
Strange, `checkid` written exactly for that problem detection.

donald2016, 2016-12-06:
I found that most of the commands are updating the target automatically, but OverrideRCIn doesn't.

TSC21, 2017-04-04:
Closed as solved

closed 2017-04-04 by TSC21