mavlink/mavros #630: Fixed Wing Offboard Ignores Attitude Setpoints

aivian, 2016-11-28:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

Testing offboard mode HiL (flightgear with custom aircraft). Starting from "stabilized" mode with mavros running and a test script commanding sinusoidal attitude and throttle commands (+/- 30 deg bank, 0-1 throttle with period 10 and 5 s respectively, using mavros/setpoint_attitude/*). I use the RC switch to change to "offboard", px4 accepts the mode switch but seems just to send me to "stabilized" mode -- inputs from the rc transmitter continue to command the vehicle.

The analyze screen makes it seem like the commands are making it to the vehicle in some form -- the throttle command appears sporadically superimposed on the manual throttle command.
[image]

If I switch to offboard from hold mode, then the commands are clearer:
[image]

Regardless of which mode I transition from, in offboard the vehicle continues to respond to the transmitter as if it were in stabilize mode.

### MAVROS version and platform

Mavros: 0.18.4
ROS: Kinetic
Ubuntu: 16.04 arm (Odroid C1+)

### Autopilot type and version

[  ] ArduPilot
[x] PX4

Version: 1.5.0

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```
[node_log.txt](https://github.com/mavlink/mavros/files/617422/node_log.txt)

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
[diagnostics.txt](https://github.com/mavlink/mavros/files/617424/diagnostics.txt)

```

### Check ID

```
rosrun mavros checkid
[checkid.txt](https://github.com/mavlink/mavros/files/617429/checkid.txt)

```

aivian, 2016-11-28:
Initial post mangled some of my logs. Retrying.
[diagnostics.txt](https://github.com/mavlink/mavros/files/617431/diagnostics.txt)
[checkid.txt](https://github.com/mavlink/mavros/files/617432/checkid.txt)

vooon, 2016-11-29:
Looks like offboard timeout. What frequency of publish rate? Recommended is 20-50 Hz.

aivian, 2016-11-29:
Both topics are published at 20Hz. I've gotten the offboard timeout before, it throws an offboard comm loss flag of some kind and goes into a failsafe. That does not happen here. If I stop the stream of offboard commands the FCU throws a flag:

`FCU: failsafe mode on`

vooon, 2016-11-30:
Looks like chain `controller -> mavros -> fcu` works.
Try other type of controls. Most tested is local control.
Then you may locate problem in firmware attitude sp handler. Ensure that you update Header.stamp at each publish.

aivian, 2016-11-30:
Ok, sounds like this is a problem that will have to be chased down on the px4 side then. Thanks for the help.

(also, going forward, did I file this in the right place? I would have taken it to the discuss.px4 channel but was having trouble with it at the time.)

vooon, 2016-11-30:
PX4 discuss should be right place.

closed 2016-11-30 by aivian