mavlink/mavros #631: PixHawk mavros Broadcast request timeout retries left 4

Tondy, 2016-12-04:
Hello

Having some issues with mavros and pixhawk.   After running

roslaunch mavros px4.launch error below appears I read around
but only came across pixhawk timestamp issue highlight. All the
time stamp blogs are not clear on step by step fixes or the issue
hope you can help

### Issue details

[ WARN] [1480855668.187286633]: VER: broadcast request timeout, retries left 4
[ WARN] [1480855669.186871175]: VER: broadcast request timeout, retries left 3
[ WARN] [1480855675.187690134]: VER: unicast request timeout, retries left 2
[ WARN] [1480855677.181476759]: PR: request list timeout, retries left 2
[ WARN] [1480855678.182365759]: PR: request list timeout, retries left 1
[ WARN] [1480855679.183363426]: PR: request list timeout, retries left 0
[ WARN] [1480855680.194696426]: VER: unicast request timeout, retries left 1
[ WARN] [1480855685.202809718]: VER: unicast request timeout, retries left 0
[ WARN] [1480855690.209882510]: VER: your FCU don't support AUTOPILOT_VERSION, switched to default capabilities

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Kinetic?
Ubuntu: ?16.04?

### Autopilot type and version

[  ] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
rosrun mavros checkid
```

vooon, 2016-12-04:
Please edit your post to fill all [template fields][1]! Provided log is not enough.

[1]: https://github.com/mavlink/mavros/blob/master/.github/ISSUE_TEMPLATE

Tondy, 2016-12-04:
hello

i just switched it on and ran the launch command to edit the form but the error is not coming back tried a few times the error below is now appearing. I can arm with the error below. please advice

[ WARN] [1480879872.663082225]: CMD: Unexpected command 520, result 0
[ERROR] [1480879881.654451765]: FCU: APM:Copter V3.3.3 (acf2e10c)
[ERROR] [1480879881.655258558]: FCU: PX4: 34e1d543 NuttX: 7c5ef883
[ERROR] [1480879881.656553243]: FCU: Frame: QUAD
[ERROR] [1480879881.657160379]: FCU: PX4v2 00320028 30345108 31323737
[ INFO] [1480879886.655334832]: WP: mission received
[ INFO] [1480879891.855688862]: PR: parameters list received

vooon, 2016-12-05:
`FCU:` messages is coming from autopilot. Looks like you have bad contact, check wiring.

Tondy, 2016-12-12:
Hello

So after switching of the odroid and pixhawk. The original error appeared again. With the fcu error I could still arm the drone . On the current error arming does not work. Hope you can help.

rosrun mavros mavsafety arm

results in

Request failed. Check mavros logs

cheers

vooon, 2016-12-12:
What printed in logs? I did not see.
Launch mavros, run arm, then copy all mavros output.

Tondy, 2016-12-12:
Hi many thanks for reply

Below is the mavros error . 

`roslaunch mavros px4.launch
... logging to /home/odroid/.ros/log/22c1eadc-c083-11e6-a4ca-ad3af56dff7a/roslaunch-odroid-5822.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://odroid:38222/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_url: /dev/ttyUSB0:921600
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: fcu
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: False
 * /mavros/local_position/tf/send_fcu: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: True
 * /mavros/mocap/use_tf: False
 * /mavros/plugin_blacklist: ['safety_area', '...
 * /mavros/plugin_whitelist: []
 * /mavros/px4flow/frame_id: px4flow
[... 241 lines cut ...]
[ INFO] [1481558322.317702210]: Built-in MAVLink package version: 2016.5.20
[ INFO] [1481558322.317802585]: Built-in MAVLink dialect: ardupilotmega
[ INFO] [1481558322.317915627]: MAVROS started. MY ID 1.240, TARGET ID 1.1
[ INFO] [1481558322.838648711]: CON: Got HEARTBEAT, connected. FCU: ArduPilotMega
[ INFO] [1481558323.854438669]: VER: 1.1: Capabilities 0x0000000000000b83
[ INFO] [1481558323.854818919]: VER: 1.1: Flight software:     030303ff (acf2e10c34e1d5437c5ef883)
[ INFO] [1481558323.855091044]: VER: 1.1: Middleware software: 00000000 (34e1d5437c5ef883)
[ INFO] [1481558323.855609252]: VER: 1.1: OS software:         00000000 (7c5ef883)
[ INFO] [1481558323.855888502]: VER: 1.1: Board hardware:      00000000
[ INFO] [1481558323.856164627]: VER: 1.1: VID/PID: 0000:0000
[ INFO] [1481558323.856425544]: VER: 1.1: UID: 0000000000000000
[ WARN] [1481558323.857061086]: CMD: Unexpected command 520, result 0
[ERROR] [1481558332.845043962]: FCU: APM:Copter V3.3.3 (acf2e10c)
[ERROR] [1481558332.845864462]: FCU: PX4: 34e1d543 NuttX: 7c5ef883
[ERROR] [1481558332.846661378]: FCU: Frame: QUAD
[ERROR] [1481558332.847301628]: FCU: PX4v2 00320028 30345108 31323737
[ INFO] [1481558337.845569796]: WP: mission received
[ INFO] [1481558343.047046630]: PR: parameters list received
`

With the above error: rosrun mavros mavsafety arm
arming works. I am looking at getting new cable please send me a link to a nice site to
source a cable for usb to pixhawk cable if you know one.

many thanks

vooon, 2016-12-13:
Why you use px4.launch with APM firmware?

Tondy, 2016-12-13:
Hello

 So I assumes when I had pixhawk connected to mission planner it had px4 firmware instead of apm. I also just started using QGroundcontrol . I don't mind what firmware really. Thank you for pointing it out. I just checked in mission planner they is only options to install apm firmware its not clear where to install px4 firmware accept for a link that says force px4 bootloader. 

In QGroundcontrol firmware options says>
"QGroundControl can upgrade the firmware on Pixhawk devices, SiK Radios and PX4 Flow Smart Cameras. All QGroundControl connections to vehicles must be disconnected prior to firmware upgrade.
Please unplug your Pixhawk and/or Radio from USB."

I have been looking at 
https://pixhawk.org/firmware/downloads

But it is not clear how px4 firmware is flashed on to pixhawk. Otherwise I will use roslaunch mavros apm.launch instead.

Many thanks

vooon, 2016-12-13:
Pixhawk (hardware) can run PX4 or APM firmwares.
I don't think that you really need px4, since apm easier to use.
But perhaps better to update to more recent version of APM Copter, 3.3.3 released year ago.

closed 2017-01-23 by vooon