mavlink/mavros #632: Sending custom mavros msg to GCS from companion computer.

annesteenbeek, 2016-12-05:
Hello,

I am running Mavros on a RPi companion computer connected to Telem2 of the FCU running arducopter.
I want to send values collected by the RPi from extra sensors to the GCS (Qgroundcontrol atm) via the telemetry connected to the FCU on Telem1.
I was planning on doing this by sending the custom mavlink message to the ~mavros/to topic and setting a target_system_id value of 0 for broadcasting ([ArdupilotDoc](http://ardupilot.org/dev/docs/mavlink-routing-in-ardupilot.html)). This way i was hoping to prevent having to add additional complexity to the FCU firmware.
However i can't find a way to add this broadcast value to the [mavros_msgs/Mavlink.msg](https://github.com/mavlink/mavros/blob/master/mavros_msgs/msg/Mavlink.msg).
And if I were to set this value using the ~target_system_id param all my messages would be broadcasted.

Is there a better way to do this or would I have to write a custom mavros plugin to achieve this or is there a another way I am missing?
Would it also be possible to add the target_id to the Mavlink.msg, would this be a nice feature?

In #574 it was suggested to send via the gcs_bridge from topic, but this would require the GCS to be connected to that bridge?

vooon, 2016-12-07:
#574 assume that telemetry flow in one direction: FCU <-> OBS <-> GCS. E.g. modem connected to computer, not autopilot.
However APM may route messages, so your approach should work.

Mavlink frame format has only source system and component id. `target_system` and component is two members of message struct.
Example: [PARAM_REQUEST_READ](http://mavlink.org/messages/common#PARAM_REQUEST_READ).

Better to write plugin. Why people so love to do more to send data?

annesteenbeek, 2016-12-08:
Thanks for the reply,

I've written a simple [plugin](https://github.com/annesteenbeek/mavros/blob/agrodrone/mavros/src/plugins/tank_level.cpp) now. Added it to the plugin list, created a new mavlink dialect which included common.xml (for some reason i can't include ardupilotmega.xml since then the build fails when i add the target_system and target_component fields. Is there a way around this since I would prefer to use an extend ardupilot dialect?

I've recompiled the mavlink with my new dialect using catkin build (still using mavlink v1.0 on Indigo) and included the plugin in the apm_pluginlist.yaml.
To test i run arducopter SITL and then have a node broadcast to the subscribed topic.
But when I try to find the broadcasted message using the Qgroundcontrol mavlink Inspector widget I can't find the message i wanted to Broadcast. 

I was hoping that i wouldnt have to include the dialect in the arducopter firmware and that it would just broadcast. Should i also recompile the GCS using the new mavlink dialect? 
How do you normaly debug mavlink? How can I know at what point in the chain the problem is?

vooon, 2016-12-08:
a) You don't have to modify mavros sources to make your plugin. mavros_extras started as example on how to make package with custom plugin.
In short you need:
1. library (.so) build from plugin.cpp
2. mavros_plugins.xml file with pluginlib description (class loader)
3. package.xml should have export entry with xml \#2
4. CMakeLists.txt should install xml and library.

b) Why you use 0.17? Switch to 0.18. That version support mavlink v1.0 too. And for me easier to support :)

c) APM and QGC should know your message, else it will be dropped.
So i believe easier to insert your messages into ardupilotmega. Also you may use include, see uAvionix.xml vendor message set.

And why not to switch v2.0?

vooon, 2016-12-08:
For my purposes common message set is enough, so i do not worry about gcs and firmware modifications.
Just check with wireshark data flow and see gcs message window.

annesteenbeek, 2016-12-10:
Currently I'm still using indigo, once i have this working i'l convert  it to a standalone plugin and also use v2.0.
I've included modified mavlink sets in qgroundcontrol and arducopter.
At first Qgroundcontrol wasnt receiving anything but turns out this was due to the fact that mavproxy doesnt act just as a proxy but also filters out messages it does not know. I havent been able also modify the mavproxy mavlink library but i've found a workaround. Now Qgroundcontrol is receiving the message.

Thanks for the help.

closed 2016-12-10 by annesteenbeek

eduardofamendes, 2018-03-12:
What workaround did you use?