mavlink/mavros #634: HIL_ACTUATOR_CONTROL is not published

mzahana, 2016-12-11:
I am run-in Pixhawk (PX4 firmware) in HIL mode. This is done by setting the HIL Quadcopter X airframe from QGC. HIL sensors are being sent from simulator (V-REP) and verified to be working properly. Now, I am interested to get the control from Pixhawk. There are two topics for this that are supposed to be published by MAVROS: ```HIL_CONTROL``` and ```HIL_ACTUATOR_CONTROL```. I can echo ```HIL_CONTROL```, but not ```HIL_ACTUATOR_CONTROL```. What could be the reason?

Thanks.

### MAVROS version and platform

Mavros: compiled from master
ROS: Indigo
Ubuntu: 14.04

### Autopilot type and version
[  ] PX4

Version: 1.5.2

vooon, 2016-12-11:
checkid.

mzahana, 2016-12-12:
Here is what I get when I run checkid

```bashrosrun mavros checkid 
OK. I got messages from 1:1.

---
Received 5508 messages, from 2 addresses
sys:comp   list of messages
 189:1     1
  1:1     0, 1, 140, 141, 147, 24, 30, 32, 33, 36, 65, 69, 70, 74, 83, 87, 91, 105, 111, 241, 242, 245
```

The [HIL_ACTUATOR_CONTROL](https://pixhawk.ethz.ch/mavlink/#HIL_ACTUATOR_CONTROLS) is not there!
However, [HIL_CONTROL](https://pixhawk.ethz.ch/mavlink/#HIL_CONTROLS) is there. Also, it's mentioned in different places that HIL_ACTUATOR_CONTROL is replacement for HIL_CONTROL!

vooon, 2016-12-12:
Did you enable that message stream? https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_messages.cpp#L2298

mzahana, 2016-12-12:
No, I didn't. How to do that?

vooon, 2016-12-12:
Not sure how, but in 2015 via extras.txt on SD.
Now better to ask http://discuss.px4.io/

closed 2017-03-03 by vooon