mavlink/mavros #637: Receiving custom message from GCS

annesteenbeek, 2016-12-17:
Hello,

I wrote a new message specification and incorperated this into the mavlink package, I then added the same message specification to Pymavlink and created a module in Mavproxy to send my new message to the FCU, the FCU is receiving my new message correctly but for some reason I can't handle the message in Mavros. (All using ardupilot SITL)

I created a custom plugin for receiving the message and am listing for it using the 
```
MESSAGE_HANDLER(MAVLINK_MSG_ID_*custom message here*, etc...
```
But nothing seems to trigger it,

I was checking the mavlink/from topic but the custom message id does not show up there either.

Is there something I am missing? I'm sure mavros is using the correct mavlink headers and everything.

vooon, 2016-12-18:
Does FCU send this message?
What version of mavros?

annesteenbeek, 2016-12-18:
No MAVproxy sends it to the FCU. Should the FCU be sending the messages for mavros to handle them? 
Then a simple ACKnolage would work.
I'm still using 0.17 (upgrading soon).

vooon, 2016-12-18:
Mavros **only** talks with FCU.

annesteenbeek, 2016-12-18:
So it only handles messages that have been send from the FCU directly to the Companion?
My plan was to send a message from mavrpoxy to the FCU which would then broadcast it so that the Companion (mavros) would receive it and be able to handle it. So am I correct in that mavros does not handle any of the broadcasted messages or could this be caused by SITL which causes mavproxy and mavros to be seen as the same channel?

My plan now is (assuming broadcasted messages are out of the question) to send a custom command using command_long to the FCU from the GCS and then have mavros intercept the COMMAND_ACK message and retrieve the command ID. This however would be less then ideal since this way I wouldn't be able to use the command parameters, just the command ID also my plugin would constantly be checking every /from message which does not seem very efficient.

The other solution would be to modify ardupilot/px4 to send out the command again to all routes once it's been received but i would prefer not to modify the fcu code since this seems kind of hackish for this purpose.

Shouldn't it be possible to set the target_component or target_system from the GCS so that mavros on the companion port would receive it instead of just the FCU itself? I just feel like there is a better/cleaner way to achieve this.

vooon, 2016-12-19:
That may work, only if FCU router will resend message to companion. Setting target may help, but i never setup that chain.
Don't know does it work or not.

annesteenbeek, 2016-12-20:
Currently using SITL the FCU is resending the message, just not to mavros, I think this is due to routing in SITL because it thinks mavproxy and mavros are sending from the same direction.
Currently I'm manually resending it, this works. 
This week I'l test everything on the hardware and see if broadcasting works then.

closed 2017-03-03 by vooon